f4exb/sdrangel #1812: Is possible avoid AVX512 in cmake?

hiperiondev, 2023-09-06:
The detection of the AVX512 flag is automatic. This causes problems when compiling a generic Copr package for Fedora.
Is there a flag to force it not to be detected automatically?

RobRich999, 2023-09-07:
Try setting the cmake build option -DARCH_OPT= to an architecture without AVX512 support.

Thing is that is also going to set a compiler baseline, so if deploying to a wide userbase, you probably need to set something like -DARCH_OPT=x86-64-v2 to force disable AVX/AVX2/AVX512 altogether.

x86-64-v2 limits SIMD to SSE 4.2, which is the required minimum SIMD support level for its own release builds anyway.

SDRangel release build cmake config stats as tested with GCC 13.1 on Linux with modern AVX2-class proc:

```
-- Target architecture: x86_64-x86-64-v2
-- Performing Test FLAG_SUPPORTED
-- Performing Test FLAG_SUPPORTED - Success
-- Looking for __ARM_NEON
-- Looking for __ARM_NEON - not found
-- Looking for __AVX512F__
-- Looking for __AVX512F__ - not found
-- Looking for __AVX2__
-- Looking for __AVX2__ - not found
-- Looking for __AVX__
-- Looking for __AVX__ - not found
-- Looking for __SSE4_2__
-- Looking for __SSE4_2__ - found
-- Looking for __SSE4_1__ - forced found
-- Looking for __SSSE3__ - forced found
-- Looking for __SSE3__ - forced found
-- Looking for __SSE2__ - forced found
```

hiperiondev, 2023-09-08:
I'll test this option

hiperiondev, 2023-09-08:
It worked perfectly!
Thanks a lot

closed 2023-09-08 by hiperiondev