f4exb/sdrangel #1817: SigMF example IQ files on site

Teque5, 2023-09-08:
I am one of the maintainers of [SigMF](https://github.com/sigmf/SigMF) and the [sigmf-python module](https://github.com/sigmf/sigmf-python) and at the GNU Radio Conference 2023 we have been playing with your software and see you have SigMF support (cool). We used it to some some M17 puzzles.

I see you have a bunch of [example IQ files](https://www.sdrangel.org/iq-files/) on your site - which is wonderful - but reading these files requires a bit of manual labor.

If I download and convert all these into clean `.sigmf` or `.sigmf.zip` files, is there someone I can deliver these file to in order to update the website? This would allow users to avoid setting samp_rate, datatype, and give you checksums and additional metadata for the IQ data.

srcejon, 2023-09-08:
They already contain sample rate, frequency etc.

Teque5, 2023-09-08:
They would still be single files (actually a `.sigmf` extension that is actually a `tar` containing a `.sigmf-data` and `.sigmf-meta` file).

It is true that your `wav` recordings have the minimum required metadata, but i'm not sure about your `sdriq` files since I've never seen that before. I only created the issue in case you wanted to adopt a more feature-complete radio specific format. Having license info, authorship, checksums, time/freq annotations, et cetera have been a leap for many of us working with these kinds of recordings all the time.

f4exb, 2023-09-10:
`.sdriq` is SDRangel specific

Teque5, 2023-09-10:
> `.sdriq` is SDRangel specific

Right. Well I'm offering to convert all the files but if you aren't interested you can close this issue.

Am I free to use those samples for my own use or redistribute(?) - there is no license stated on the site (something else that would be in a SigMF metadata) - not trolling, genuinely curious.

closed 2023-10-26 by github-actions[bot]