f4exb/sdrangel #1818: fm ts offset

RichNeese, 2023-09-09:
### way on the TX T:1 to setf a +/- offset for FM repeaters  and the wifi does not discuss it that I can find I have read the full on wiki

f4exb, 2023-09-10:
can you please rephrase it in proper English?

RichNeese, 2023-09-11:
As a deaf ham . English is not my first language and American English is not my 2nd.

The question is … How do you set the offset for fm repeater . Nothing I find int eh wiki discusses this .

f4exb, 2023-09-11:
Thanks this is much clearer. There is no such thing like that. Rx and Tx
are separated thus you have to set frequencies separately. In the case the
center frequencies of both devices are linked and equal you have to get a
large enough bandwidth to be able to separate the Rx and Tx channels wide
apart enough.

f4exb, 2023-09-17:
Please follow comments above...

closed 2023-09-17 by f4exb