f4exb/sdrangel #1819: Audio input does not work on MacOs versions.

rkarikari, 2023-09-10:
Hello,

I'd like to bring the following to your attention.
 
Rx device "Audio"  does not work until the following lines are included ... at the end of  the  info.plist file.

```
<key>NSMicrophoneUsageDescription</key> 
<string>This app requires microphone access to receive signals</string>
```

I have had  to make this edit on every version upgrade since  v 14.x 
I'm surprised no one else has mentioned this.

My OS version is Monterey 12.6.8

Sometimes I need to reset as well ....

`tccutil reset Microphone com.f4exb.SDRangel`

                                               .... to get it working .
Thank you so much .

srcejon, 2023-09-11:
It seems the problem was reported on the mailing list sometime ago: https://groups.io/g/sdrangel/topic/74229711#

Just trying it here, however, I have the same problem as Fabio, in that the above doesn't help - Microphone permission is only granted if SDRangel is run from the terminal, not from Finder. However, I'll add a patch to add that key in to Info.plist anyway, as it seems to be required. I'll also add camera and location.

rkarikari, 2023-09-11:
Thanks so much for looking into this.
Much appreciated.
Thank you.

PR #1822 (2023-09-11): Add description why Microphone, Camera and Location are required on MacOS.
In Info.plist to help fix #1819
closed 2023-09-11 by f4exb