f4exb/sdrangel #1820: BUG: On Raspberry Pi 4B with 8GB of RAM, build fails if USRP included

gvimlag, 2023-09-10:
PROBLEM:
  On a Raspberry Pi 4B with 8GB of RAM; if I add the USRP hardware dependency, then the SDRangle build fails.
  But on an x86_64 CPU/OS; if I add the USRP hardware dependency, then the SDRangle build succeeds.
  The SDRangel/USRP build only fails on the Raspberry Pi.

SYSTEM:
  Hardware: Raspberry Pi 4B (8GB RAM)
  OS:       Raspberry Pi OS (bullseye) 64-bit

  SDRangel: (compiled from source following "https://github.com/f4exb/sdrangel/wiki/Compile-from-source-in-Linux")
    Version 7.15.4-47-gffbb26c36
    Build info: Qt 5.15.2 64 bits
    DSP Rx 16 bits    Tx 16 bits

Build output from Raspberry Pi 4B with 8GB RAM:
...
Scanning dependencies of target sdrangelbench_autogen
[100%] Automatic MOC and UIC for target sdrangelbench
[100%] Built target sdrangelbench_autogen
Scanning dependencies of target sdrangelbench
[100%] Building CXX object CMakeFiles/sdrangelbench.dir/sdrangelbench_autogen/mocs_compilation.cpp.o
[100%] Building CXX object CMakeFiles/sdrangelbench.dir/appbench/main.cpp.o
Assembler messages:
Fatal error: can't create CMakeFiles/outputusrp.dir/outputusrp_autogen/mocs_compilation.cpp.o: Bad message
make[2]: *** [plugins/samplesink/usrpoutput/CMakeFiles/outputusrp.dir/build.make:82: plugins/samplesink/usrpoutput/CMakeFiles/outputusrp.dir/outputusrp_autogen/mocs_compilation.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:14858: plugins/samplesink/usrpoutput/CMakeFiles/outputusrp.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[100%] Linking CXX executable sdrangelbench
[100%] Linking CXX shared library ../../../lib/plugins/liboutputplutosdr.so
[100%] Built target sdrangelbench
[100%] Linking CXX shared library ../../../lib/plugins/liboutputsoapysdr.so
[100%] Built target outputplutosdr
[100%] Built target outputsoapysdr
make: *** [Makefile:171: all] Error 2

srcejon, 2023-09-11:
Are you doing a parallel make? (i.e. with -j N) If so, don't use -j.

gvimlag, 2023-09-13:
Hello srcejon,

`Are you doing a parallel make? (i.e. with -j N) If so, don't use -j.`

As you suggested, I rebuilt using:
```
#make -j $(nproc)    #FIX: remove "PARALLEL make" ("PARALLEL make" causes build to fail)
 make                #FIX: remove "PARALLEL make" ("PARALLEL make" causes build to fail)
```

Your suggestion worked!
THANK YOU!

One of the possibilities of why your suggestion worked is that it freed up some total memory.
So I did some additional testing to see if increasing the total memory would solve this build problem WITHOUT modifying the build script.

The DEFAULT total memory on this Raspberry Pi is 8.1GB (8GB of RAM plus 0.1GB of SWAP).
I decided to increase the SWAP size from 0.1GB to 1GB, thereby raising the total memory from 8.1GB to 9GB.
This total memory increase solved the build problem WITHOUT modifying the build script.
So the Raspberry Pi needed more than 8.1GB of total memory to build this script.

If possible, could you put a note in the build script warning people of this total memory issue when building on a Raspberry Pi?

Again, thank you for your suggestion.
Much appreciated.

closed 2023-10-29 by github-actions[bot]