f4exb/sdrangel #1821: Swagger file generation appears incorrect

srcejon, 2023-09-11:
If I run the swagger generate.sh script to generate the .cpp/.h files, I see a problem in the generated:

 swagger/sdrangel/code/qt5/client/SWGRTTYModActions.h

It appears to contain references to PSK31ModActions. I can't see why this is happening though. There's no mention of PSK31 in RTTYMod.yaml and none to RTTYMod in PSK31.yaml. ChannelActions.yaml appears to point to the correct files.

Do you see the same problem?

f4exb, 2023-09-11:
If I look at the present code on master I cannot see it: https://github.com/search?q=repo%3Af4exb%2Fsdrangel%20PSK31ModActions&type=code

srcejon, 2023-09-11:
Yeah, I didn't check it in, as it wouldn't compile. So you'd need to run generate yourself.

f4exb, 2023-09-13:
It seems the generator is confused by having the "payload" section exactly the same. If I add a slightly different description for the "text" property in both .yaml files then the PSK31ModActions disappears in the RTTY mod files. While this is probably a bug in the generator it is simpler to have a different description,

commit af916d6e (2023-09-13): Fixed Swagger files generation and added missing files. Fixes #1821
closed 2023-09-13 by f4exb