f4exb/sdrangel #1823: How to configure a simple local sdrangel server/client combination?

gvimlag, 2023-09-12:
I'm trying to configure a simple local sdrangel server/client combination using a Raspberry Pi server and an X86/Debian client.
Unfortunately, I can't get a connection between the client and server (Client device window message is always: "Not connected").

Configuration:
  <Both sdrangel client and server compiled from source following "https://github.com/f4exb/sdrangel/wiki/Compile-from-source-in-Linux">

  SDR Device:
`    SDRplay RSP1A connected to Raspberry Pi 4B server`

  sdrangel Server:
```
    Raspberry Pi 4B with 8GB RAM
    full 64-bit Raspberry Pi OS
```

  sdrangel Client:
```
    X86_64 with 8GB RAM
    Debian 11.7 64-bit OS
```

On the Raspberry Pi server:
`  /opt/install/sdrangel/bin/sdrangelsrv    #using all defaults`

On the x86_64 client:
```
  /opt/install/sdrangel/bin/sdrangel   #using all defaults
    Add Rx device->RemoteTCPInput
      IP: <IP address of Raspberry Pi> : 1234    #default port address
  Device window message: "Not connected"
```

I've searched the documentation:
```
  https://github.com/f4exb/sdrangel/wiki/SDRangel-server
  https://www.google.com/search?client=firefox-b-1-e&q=%22sdrangel%22+%22server%22#fpstate=ive&vld=cid:f67fe033,vid:sKiEvA1nejc,st:0
```

I verified that the SDRplay RSP1A device is recognized by the server:
```
  sdrangelsrv --list-devices
    Available devices:
    HWType: AudioInput Serial: 0
    HWType: SDRplayV3 Serial: <serial number>
```

I've tried forcing the server to use the SDRplay hardware:
`  sdrangelsrv --remote-tcp-hwtype SDRplayV3`

I've tried different ports in RemoteTCPInput: 1234, 8091, 9091, etc.

I verified that the web interface connects using a web browser: 
`  http://127.0.0.1:8091    #Developer information`

But I can't get the client to connect to the server:
`  Client device window message: "Not connected"`

How do I configure a simple local sdrangel server/client combination?

srcejon, 2023-09-12:
You need the --remote-tcp option to sdrangelsrv. 

(You'll also need the --remote-tcp-hwtype SDRplayV3 options and possibly the --remote-tcp-address as well)

gvimlag, 2023-09-13:
Hello srcejon,
Thank you for the information.
With it, I was able to get the server functioning (but with some bugs -- I'll get back to you on those).

I see that last week you added 4 useful options to the server.  Good job.

Is there a "quick-start" guide showing how to configure the sdrangel server?
(Some examples of common configurations would be helpful to the new user.)

Thank you.

c3kkos, 2023-10-11:
I'm using sdrdaemon in raspberry pi and me too having no luck in connecting to it.

I want to compile only the sdrangelsrv for rpi.. how can i do that?

compiling all sdrangel git is insane on a rpi3... it takes weeks

srcejon, 2023-10-11:
> I want to compile only the sdrangelsrv for rpi.. how can i do that?

cmake -DBUILD_GUI=OFF -DBUILD_SERVER=ON ..

c3kkos, 2023-10-11:
Thank you for quick answer. For me to learn, and not asking anymore, in future, where i can find all this compile-time parameters?