f4exb/sdrangel #1825: Still no SDRplay devices are shown

Bra-Ger, 2023-09-12:
Hello,
Sorry, but I still have issues with SDRplay devices because they are not shown. I used the following setup.

- MacBook Pro M1 Ventura 13.5.2 
- SDRplay API 3.12 !!! (there is no 3.7 for Macs anymore)
- The following versions are not showing SDRply devices (picture attached):
7.13.xx
7.14.xx
7.15.xx

The  RSP1A device is running fine with SDRconnect from SDRplay.
There is no other SDR SW running.
I attached a Log file (I hope it contains some useful information.)

Please comment I you need more details.

Regards,

Gero

[image]
[NO SDRplay.txt](https://github.com/f4exb/sdrangel/files/12588795/NO.SDRplay.txt)

srcejon, 2023-09-12:
In your log file, you have:

Referenced from: <57595EC3-AC46-3AA8-AE6F-8C1DB9B4E978> /Applications/SDRangel.app/Contents/Resources/lib/plugins/libinputsdrplayv3.dylib
  Reason: tried: 'libsdrplay_api.so.3.07' (no such file), '/System/Volumes/Preboot/Cryptexes/OSlibsdrplay_api.so.3.07' (no such file), 'libsdrplay_api.so.3.07' (no such file), '/usr/local/lib/libsdrplay_api.so.3.07' (no such file), '/usr/lib/libsdrplay_api.so.3.07' (no such file, not in dyld cache), '//libsdrplay_api.so.3.07' (no such file), '/System/Volumes/Preboot/Cryptexes/OS//libsdrplay_api.so.3.07' (no such file), '//libsdrplay_api.so.3.07' (no such file), '/usr/local/lib/libsdrplay_api.so.3.07' (no such file), '/usr/lib/libsdrplay_api.so.3.07' (no such file, not in dyld cach

Bra-Ger, 2023-09-12:
Hello, 
Thank you for your fast response.

I already mentioned that SDRplay only provides the 3.12 API for MacOS in their Download section. 

Because this is a new instalation I can no longer a download the 3.07 Version for Mac. 

Is anybody out there who can confirm this behaviour with the 3.12 API for MacOS?
Is there anybody who can provide me a 3.07 API for M1 MacOS in order to confirm this issue?
(I will try to contact SDRplay by my self but I don't think that they will provide any support for 3rd party apps.)

(Please take a look at the attachment.)
Regards,

Gero 

[image]

srcejon, 2023-09-12:
Yep - I may need to update the build to use 3.12.

srcejon, 2023-09-13:
Which isn't straightforward, as libsdrplay_api.so.3.12.0  refers to libsdrplay_api_x86_64.so.3.12.0 which doesn't exist.

Bra-Ger, 2023-09-13:
Strange!

libsdrplay_api.so.3 -> /usr/local/lib/libsdrplay_api.so.3.12.0 
Does exist ...
Dit you try it by yourself? Maybe I should try it on a vanilla MacOS

srcejon, 2023-09-13:
I'm not referring to the symlinks, but a string within the file, which you can see with otool. I may have a workaround, but have also filed a bug report.

cygeus, 2023-09-14:
> I already mentioned that SDRplay only provides the 3.12 API for MacOS in their Download section.
> 
> Because this is a new instalation I can no longer a download the 3.07 Version for Mac.
> 
> Is there anybody who can provide me a 3.07 API for M1 MacOS in order to confirm this issue?

SDRplay API 3.07 for M1 Macs is still available for download from the website:
[https://www.sdrplay.com/software/SDRplay_RSP_API-MacOS-3.07.a4.run](https://www.sdrplay.com/software/SDRplay_RSP_API-MacOS-3.07.a4.run)
[https://www.sdrplay.com/docs/M1_Mac_API_3.07_BETA.pdf](https://www.sdrplay.com/docs/M1_Mac_API_3.07_BETA.pdf)

srcejon, 2023-09-14:
Great - I'd suggest using that. SDRplay have replied saying they are working on a fix for the 3.12.0 issue.

Bra-Ger, 2023-09-15:
I have tried the API 3.07 version but the device is not detected by this API:

SDR % sudo SoapySDRutil --probe
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Probe device 
[INFO] Make connection: ''
Error probing device: Failed to make connection with ''
gero@Geros-MBP-Max SDR % 

The is a bug in this beta which was fixed with 3.12 unfortunately this workaround is not an option.

Bra-Ger, 2023-12-12:
Thank you!!!!

At first glance it seems to be working fine. I will test it this weekend (16/17 Dec. 23) with two devices at the same time. 

 
[image]
[image]

Bra-Ger, 2023-12-17:
Hello,
I have now tested the API with two devices at the same time. 
At first everything seemed to be OK. After a while, the second device could no longer be addressed.

There was a different behavior when they were connected in different order.

In log "20231217_two_SDRplay_Crash"
SDRAngel crashes:
Line 142, 143 (the crash is no longer logged)

In log "20231217_two_SDRplay_Crash2"
SDRAngel does not crash but the API cannot be opened:
Line 102, 103

I have reset the API with
$ sudo bash /Library/SDRplayAPI/3.12.1/uninstall.sh
and deleted and reinstalled SDRAngel
(Including all .ips .diag .plist files and the folder with xxx. savedState)

Then I installed everything again. => Reboot => Test
After that I saw the same behavior:

In the log "20231217_two_SDRplay_Crash3"
SDRAngel crashes:
Line 77, 78

Conclusion: 
With MacBook Pro Mxs M114.0 (Sonoma 14.0),
SDRplayAPI 3.12.1
SDRAngle 7.17.2
operation with an SDR Play device is possible. 
Two seem to work but the function is not guaranteed.  

Translated with DeepL.com (free version)
[20231217_two_SDRplay_Crash3.txt](https://github.com/f4exb/sdrangel/files/13696319/20231217_two_SDRplay_Crash3.txt)
[20231217_two_SDRplay_Crash2.txt](https://github.com/f4exb/sdrangel/files/13696320/20231217_two_SDRplay_Crash2.txt)
[20231217_two_SDRplay_Crash.txt](https://github.com/f4exb/sdrangel/files/13696321/20231217_two_SDRplay_Crash.txt)

srcejon, 2024-02-23:
Please let us know if you still have a problem with the latest release and API 3.14.

Bra-Ger, 2024-02-24:
Hello,
Thank you for your question. I still have the same issue. 
Because SDRangle crashes so fast that no log file is written  I copied the Apple crash report. 
[2024-02-24_SDR_play_Crash.txt](https://github.com/f4exb/sdrangel/files/14392267/2024-02-24_SDR_play_Crash.txt)

srcejon, 2024-02-24:
Try running from the terminal, then the log should displayed there.

srcejon, 2024-02-24:
I see one possible bug from your crash report, so I'll fix that, thanks.

srcejon, 2025-01-25:
@Bra-Ger is this still available problem?

Bra-Ger, 2025-01-25:
Thank you for asking but unfortunately the issue still exists. I just tried it on a vanilla system. But this time it was a Win 11 and not a Mac OS 15.2. Which is strange because it used to work on win 10. The behaviour is still the same it is possible to add at the 2nd device but then it crashes. Would it be any helpful to send new crash reports from the win 11 device?

srcejon, 2025-01-26:
Ah, just a problem with the second device? According to this: https://github.com/f4exb/sdrangel/issues/2346 SDRplay API doesn't support two devivces.