f4exb/sdrangel #1829: macos crash on adding atv mod on tx

iicedancer, 2023-09-15:
i tried to add the ATV modulator with my hackrf but i get a crash instead. i'm no programmer, but this crash seems simple enough to fix?

-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               SDRangel [10828]
Path:                  /Applications/SDRangel.app/Contents/MacOS/sdrangel
Identifier:            SDRangel
Version:               ???
Code Type:             X86-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2023-09-14 20:51:14.2874 -0500
OS Version:            macOS 13.5.2 (22G91)
Report Version:        12
Bridge OS Version:     7.6 (20P6072)
Anonymous UUID:        6BE2C287-399F-2C18-98FD-0C27070554F1

Sleep/Wake UUID:       6D731838-128E-409B-81A8-B03008CB0509

Time Awake Since Boot: 120000 seconds
Time Since Wake:       21193 seconds

System Integrity Protection: enabled

Crashed Thread:        41  Dispatch queue: com.apple.root.default-qos

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000

Termination Reason:    Namespace TCC, Code 0 
This app has crashed because it attempted to access privacy-sensitive data without a usage description. The app's Info.plist must contain an NSCameraUsageDescription key with a string value explaining to the user how the app uses this data.

_(this is where the real error message was, but it was too long for a github post)_ 
 
Model: MacBookAir9,1, BootROM 1968.140.7.0.0 (iBridge: 20.16.6072.0.0,0), 4 processors, Quad-Core Intel Core i5, 1.1 GHz, 8 GB, SMC 
Graphics: Intel Iris Plus Graphics, Intel Iris Plus Graphics, Built-In
Display: Color LCD, 2560 x 1600 Retina, Main, MirrorOff, Online
Memory Module: BANK 0/ChannelA-DIMM0, 4 GB, LPDDR4X, 3733 MHz, Micron, MT53E512M64D4NW-053
Memory Module: BANK 2/ChannelB-DIMM0, 4 GB, LPDDR4X, 3733 MHz, Micron, MT53E512M64D4NW-053
AirPort: spairport_wireless_card_type_wifi (0x14E4, 0x870), wl0: Jan 14 2023 01:22:18 version 16.20.357.3.3.6.118 FWID 01-eaad3d99
Bluetooth: Version (null), 0 services, 0 devices, 0 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
USB Device: USB31Bus
USB Device: USB31Bus
USB Device: HackRF One
USB Device: T2Bus
USB Device: Touch Bar Backlight
USB Device: Apple Internal Keyboard / Trackpad
USB Device: Headset
USB Device: Ambient Light Sensor
USB Device: FaceTime HD Camera (Built-in)
USB Device: Apple T2 Controller
Thunderbolt Bus: MacBook Air, Apple Inc., 86.0

srcejon, 2023-09-15:
What version of SDRangel?

A patch I made a couple of days ago may help with this: https://github.com/f4exb/sdrangel/pull/1822 and https://github.com/f4exb/sdrangel/pull/1824

iicedancer, 2023-09-22:
sdrangel version 7.15.4
i have not compiled from source, i got the program from the dmg in releases

srcejon, 2023-09-22:
Please give 7.16 a try

closed 2023-10-29 by srcejon