f4exb/sdrangel #1831: No APT decoder in flatpak for Mxlinux libretto (Debian Bookworm)

500paolo, 2023-09-15:
Hi, in my Mxlinux the only way to install Sdrangel is flatpak. In the previous version of Mxlinux I not used flatpak because I can use the source folder and run a binary file, but now if I do the same procedure it crash instantly. In flatpak version I can find all (adb-s decode & co.) except Apt decoder for meteo sats, what I can do?

srcejon, 2023-09-15:
Compile it yourself: https://github.com/f4exb/sdrangel/wiki/Compile-from-source-in-Linux

f4exb, 2023-09-16:
Flatpack, Snap and the like are not supported so the best advice is just up there...

closed 2023-09-16 by f4exb

500paolo, 2023-09-16:
Resolved simply in this way: opened the .deb archive in th sources with Ark and copied libdemodapt.so in  /var/lib/flatpak/app/org.sdrangel.SDRangel/x86_64/stable/b106008248eb531f0e5ad5ef6e5aef53c3cb0e69e2a17db1eb2a225fd2682734/files/lib/sdrangel/plugins/. Thank you anyway...