f4exb/sdrangel #1832: SDRAngel crash during APT demod

mfarrukh90, 2023-09-15:
[sdrangel1.log](https://github.com/f4exb/sdrangel/files/12619872/sdrangel1.log)

During reception of NOAA APT, SDRAngel crashed. I have attached the logs. 
SDRAngel release 7.15.0 running on Windows 11

mfarrukh90, 2023-09-17:
This is very frequent. It crashed again during reception of NOAA APT. It is a great piece of software. I hope to find a fix for it. I can share more info if needed. thanks

[sdrangel(1).log](https://github.com/f4exb/sdrangel/files/12642528/sdrangel.1.log)

srcejon, 2023-09-17:
You have lots of messages like:

2023-09-15 20:00:04.672 (W) SatelliteTracker::networkManagerFinished:  error( 1 ):  QNetworkReply::ConnectionRefusedError :  "Connection refused"

2023-09-15 20:00:12.152 (D) HttpConnectionHandler (0x27a8561b780): received request from ::ffff:127.0.0.1 (GET) /sdrangel
2023-09-15 20:00:12.155 (D) HttpConnectionHandler (0x27a8561b780): disconnected

Do you have the firewall blocking sdrangel?

mfarrukh90, 2023-09-19:
There was an antivirus that I have now uninstalled. As a result,  the crash is not frequent now. However, it does crash when APT reception and decoding starts not every time though. I am trying to identify the scenarios when it actually happens. Will share the logs soon.
Thanks

mfarrukh90, 2023-10-01:
Apparently there is some memory problem with maps feature of APT demod, RAM usage shoots up to 4GB when left running for longer periods of time. We tried running APT Demod without maps feature and observed its RAM usage, the problem seems to be under control (RAM consumption ~500MB running in idle state).

srcejon, 2023-10-01:
The APT Demod will send a received image to the map for display. If you leave it running, these images will be overlaid on top of each other, and so will use more memory for each image, so I'd expect to go up (although haven't looked at by how much). There is a button in the APT Demod GUI to delete images that are displayed on the map. Does that help reduce memory use?

I guess we could have an option not to send images to the map, in case you want to leave the map open.

closed 2023-11-16 by github-actions[bot]