f4exb/sdrangel #1833: CSV file not writing

doftheworld, 2023-09-20:
I am having an issue where the CSV file only writes when the option is turned on, turned off, or I close SDRangel. When I do any of those action, everything waiting is written to the file.  The CSV file shows that it is open by SDRangle, just doesn't get updated until one of the actions previously mentioned are performed.

srcejon, 2023-09-20:
This is probably to be expected, as is usually the case for buffered files. It should also be written when the buffer is fills up. What plugin are you using? Why do you need the file contents sooner?

doftheworld, 2023-09-20:
> This is probably to be expected, as is usually the case for buffered files. It should also be written when the buffer fills up. What plugin are you using? Why do you need the file contents sooner? What OS?

This is on windows. The plugin is the Pager Demodulator (I actually thought I was commenting on that plugin's thread). I am using Windows 10. I need the contents almost immediately as the data received includes coordinates that are being plotted on a real-time map. I can use the UDP option, however the CSV file is easier to work with and I already had a Python script that could use the data I needed.

srcejon, 2023-09-20:
Above patch should flush the log after each message.

What format are the coordinates in your messages? I could probably display them on the SDRangel map.

doftheworld, 2023-09-21:
> Above patch should flush the log after each message.
> 
> What format are the coordinates in your messages? I could probably display them on the SDRangel map.

Wow thank you so much for the quick patch!  I am on a plane at the moment, so I will try it out once I am on the ground and settled. The coordinates show up in the message as LAT:39.12829 LONG:-86.18529. - but don't put any effort into having it show up in the map. I need to be able to access them remotely so having them there wouldn't really be helpful, but thank you for the thought. If you would like, I can share with you the project I'm working on.

doftheworld, 2023-09-22:
> Above patch should flush the log after each message.
> 
> What format are the coordinates in your messages? I could probably display them on the SDRangel map.

I went to try this out and there isn't a windows version compiled (that I see). I lack the knowledge on compiling source code (although it would be nice to eventually learn). Do you by chance have a compiled version for Windows available?  Thanks!

srcejon, 2023-09-22:
There's a problem with the Windows build at the moment, so will need to wait for that to be fixed first. But when it is, it will appear on the Releases page.

doftheworld, 2023-09-22:
Okay no problem. I actually took the leap and tried to compile it myself,
but kept getting errors. I will just wait it out then - thank you!

srcejon, 2023-09-22:
Windows build instructions are here: https://github.com/f4exb/sdrangel/wiki/Compile-in-Windows

closed 2023-11-08 by github-actions[bot]