f4exb/sdrangel #1835: Window positions in MDO area of Workspace weirdness

lrowen, 2023-09-21:
So for a long time now I've had a workspace set up with several plugins loaded.  Just recently, SDRangel has stopped remembering the positions at close and has started up with the plugin windows far to the right, off the current visible section  of the workspace, and have to be manually moved back into place on every startup.  I made a short screencast video of the effect; my setup is a 4-monitor one, and this video is of the monitor on which I run SDRangel.  If you want a video of the full screen setup (a T-shape, three monitors across with the fourth monitor below the center top one) I can record that and link as well.  Here's a link to the screencast video:
[https://nebula.pari.edu/index.php/s/iFo9rftbDX2wDz5](https://nebula.pari.edu/index.php/s/iFo9rftbDX2wDz5)

As can be seen in the video, even after I save the configurations this occurs.  So I have to reconfigure the window positions every time I restart SDRangel now, which I didn't have to do prior to a few weeks ago.

Maybe I'm missing something, but I've not found a separate 'save Window positions' option.  

CHANGELOG shows version 7.15.4-1 but I track my fork in git; my git revision is:
```
ops@ops-sdr1:/opt/build/sdrangel$ git show
commit f6ea6f8a2e51c2a4edd3cee5e98808f6bff1ac04 (HEAD -> master, origin/master, origin/HEAD)
Merge: a0c75f040 8afa83c01
Author: Edouard Griffiths <f4exb06@gmail.com>
Date:   Sat Sep 16 10:30:40 2023 +0200

    Merge pull request #1830 from srcejon/fix_1827
    
    Remote TCP improvements

ops@ops-sdr1:/opt/build/sdrangel$ 

```

srcejon, 2023-09-21:
I don't recall any recent changes to this code - it should save positions automatically. 

What happens if you close all windows and workspaces, exit SDRangel and restart, then recreate your workspace?

lrowen, 2023-09-21:
I'll give that a try tomorrow as I set up for a pass.

lrowen, 2023-09-21:
Ok, some solid data here.

Using my laptop, since I can reproduce the issue there, I shut it down, undocked, and started up in single monitor mode.  SDRangel restores the positions ok, although they're not in exactly the same position.

Redocking, I found the position the plugin windows go to to be dependent upon the monitor the SDRangel window is on.  If it's the leftmost monitor, the position restores correctly.  If it's the middle monitor, the position restores somewhat to the right.  If it's on the rightmost monitor, the position restores much farther to the right.
Single monitor screencast: [https://nebula.pari.edu/index.php/s/8HbbiBLSnffkskb](https://nebula.pari.edu/index.php/s/8HbbiBLSnffkskb)

Triple monitor screencast: [https://nebula.pari.edu/index.php/s/Dg78LcPzAzCptjB](https://nebula.pari.edu/index.php/s/Dg78LcPzAzCptjB)

In hindishgt, the issue first presented itself when I went from three to four monitors in OPS, with SDRangel on the middle monitor, which had been the leftmost monitor.  Aha.  Are the plugin windows' positions saved as offsets from the SDRangel workspace origin, or from the desktop origin?  And then restored as offsets from the workspace origin?

lrowen, 2023-09-25:
NOTE: This appears to possibly be a QT5 bug in QtWidget::restoreGeometry when used in a multi-monitor configuration and MDI SubWindows.  It appears to be a rather old bug, and I don't see anywhere showing it's fixed.  The documentation for QWidget::restoreGeometry states it's used for 'toplevel' geometry, which is exactly the effect that I am seeing, and documented at:
[https://www.qtcentre.org/threads/39529-QMdiSubWindow-saveGeometry-restoreGeometry](https://www.qtcentre.org/threads/39529-QMdiSubWindow-saveGeometry-restoreGeometry)

There are some workarounds in several of the bug reports from way back, including one that completely reimplemnted restoreGeometry for MDI subwindows to base the coordinates off of the workspace and not the full desktop.

[https://bugreports.qt.io/browse/QTBUG-18648](https://bugreports.qt.io/browse/QTBUG-18648)
[https://bugreports.qt.io/browse/QTBUG-1645](https://bugreports.qt.io/browse/QTBUG-1645)

VERY OLD bug reports.

srcejon, 2023-09-26:
Will try to find an extra monitor somewhere, so I can take a look.

srcejon, 2023-11-14:
Also a problem on Windows, with both Qt 5 and 6.

As mentioned above, comment for QWidget::saveGeometry says it is for top-level widgets, and QMdiSubWindow isn't a top-level widget (as they have a parent).

Should be possible to roll our own version of saveGeometry anyway, as the code seems fairly simple:

For Qt 5.15 to 6.6, we have:

    quint16 majorVersion = 3;
    quint16 minorVersion = 0;
    const int screenNumber = QGuiApplication::screens().indexOf(screen());
    stream << magicNumber
           << majorVersion
           << minorVersion
           << frameGeometry()
           << normalGeometry()
           << qint32(screenNumber)
           << quint8(windowState() & Qt::WindowMaximized)
           << quint8(windowState() & Qt::WindowFullScreen)
           << qint32(screen()->geometry().width()) // added in 2.0
           << geometry(); // added in 3.0

srcejon, 2023-11-14:
There's also a bug on Linux on single monitor setups, whereby windows aren't restored to their previous position anyway. They appear about 40 pixels lower than they should be.

srcejon, 2023-12-01:
Should be fixed in 7.17.1.

closed 2023-12-01 by srcejon