f4exb/sdrangel #1839: NSIS issue Win 11

m6mal, 2023-09-22:
Tried to install on my 17-1265 laptop and my i7-12700 pc and it throws up "installer integrity check has failed, common causes include incomplete download and damaged media. Contact the installer's author to obtain anew copy"  and will not install at all this is reffering to the version i cant install is V7.16.0 and is only 57Mb and the last version 7.15.4 is 146MB

f4exb, 2023-09-22:
The file size starts to drop signifcantly from there: https://github.com/f4exb/sdrangel/actions/runs/6196072083

m6mal, 2023-09-22:
So why cant mine install  yesterdays version v7.16 as i get the error NSIS issue

m6mal, 2023-09-22:
just see you have reupped it in downloads and still getting instal error nsis

closed 2023-09-25 by srcejon