f4exb/sdrangel #1843: Cannot open USRP B205mini with "path not found"

sjmurdoch, 2023-10-01:
I was able to use SDRAngel with my USRP B205mini, but for reasons I am not able to work out, this stopped working. The error message I see in logs is as follows:

```
2023-10-02 00:36:58.464 (D) DeviceEnumerator::enumerateDevices: sdrangel.samplesource.usrp
2023-10-02 00:37:01.070 (D) DeviceUSRP::enumOriginDevices: found USRP device  "B205i[0:$1] 3293E94"
2023-10-02 00:37:01.070 (D) DeviceUSRPParams::open: type=b200,name=B205i,serial=3293E94,product=B205mini
2023-10-02 00:37:02.599 (D) DeviceUSRPParams::open: exception:  LookupError: IndexError: multi_usrp::mb_root(1737017948160) - LookupError: IndexError: multi_usrp::mb_root(1737017948160) - path not found
```

The USRP command line programs still work so the device seems to be available, but does not appear in SDRAngel

The version information from the logs is SDRangel 7.16.0-31-gea352d130 Qt 5.15.2 64b x86_64 Windows 10 Version 2009 DSP Rx:24b Tx:16b PID 17224

sjmurdoch, 2023-10-02:
I was able to get some logging from UHD, which seems to help

```
2023-Oct-02 01:06:39.948892,0x31596,log.cpp:505,2,UHD,Win32; Microsoft Visual C++ version 1925; Boost_107000; UHD_4.2.0.0-release
2023-Oct-02 01:06:40.041881,0x27044,b200_impl.cpp:203,3,B200,EnvironmentError: IOError: Could not find path for image: usrp_b200_fw.hex

Using images directory: <no images directory located>

Set the environment variable 'UHD_IMAGES_DIR' appropriately or follow the below instructions to download the images package.

As an Administrator, please run:

"c:/Program Files\SDRangel\uhd\utils\uhd_images_downloader.py"
```

Setting UHD_IMAGES_DIR did fix the problem, but I'm still not sure why this was necessary.

srcejon, 2023-10-02:
Do you have UHD_PKG_PATH set?

sjmurdoch, 2023-10-02:
No, that is not set at the command prompt (nor are any other UHD environment variables). Which application should have set those?

srcejon, 2023-10-02:
If the UHD installer doesn't do it, I guess you need to do it manually. 

At least one needs to be set, so it can find the FPGA firmware.

sjmurdoch, 2023-10-02:
I re-installed the UHD drivers and the `UHD_IMAGES_DIR` environment variable came back. I have no idea what could have caused it to be removed. Thanks for your help tracking this down (and for your work on SDRAngel).

closed 2023-10-11 by srcejon