f4exb/sdrangel #1844: BUGS: SDRangel RigCtl server

gvimlag, 2023-10-02:
SYSTEM:
```
  Hardware-0: x86 64-bit
  OS-0:       Debian 11.7 (bullseye) 64-bit
  SDR:        RTL-SDR V3
  GPredict:   2.3-72-gc596101-3
  SDRangel: (compiled from source following "https://github.com/f4exb/sdrangel/wiki/Compile-from-source-in-Linux")
    Version 7.16.0-33-g88aded6e0
    Build info: Qt 5.15.2 64 bits
    DSP Rx 24 bits    Tx 16 bits
```

BUGS/ISSUES:
1) Documentation for the "RigCtl server plugin" is incorrect
   (https://github.com/f4exb/sdrangel/blob/master/plugins/feature/rigctlserver/readme.md):
     Incorrect: 'Run SDRangel and from the Preferences menu select rigctrl.'
     Corrected: 'Run SDRangel and from the "Add features" menu select rigctrl.'

2) Unable to maintain connection between SDRangel and Gpredict.
   There appears to be an issue with the SDRangel "channel" code.
   (Gpredit/rigctl maintains a connection with SDR++ and GQRX; but looses connection the SDRangel.  So the problem is not with Gpredict.)

TO REPRODUCE THE CONNECTION PROBLEM BETWEEN SDRangel AND Gpredict:
```
<open terminal-1>
  rm -rf ~/.config/f4exb ~/.local/share/f4exb    #clean initial configuration
  /opt/install/sdrangel/bin/sdrangel
    Add Rx device
      RTL-SDR[0] <serial string>
      OK

    Add features
      RigCtl Server
      Apply
      Close
    [Check] Enable rigclt server
    Device:               R0
    Channel:              <BLANK>    <- UNABLE TO MODIFY "Channel"
    RigCtl Port:          4532
    Max Frequency Offset: 10000
    start/stop acquisition
```

ISSUE: "Channel:" IS EMPTY AND CAN *NOT* BE MODIFIED.

```
<open terminal-2>
  gpredict
    Edit->Preferences
      Interfaces
        Radios
          Add new
            Name:        SDRangel-RigCtl
            Host:        localhost
            Port:        4532
            Radio type:  RX only
            PTT status:  None
            VFO Up/Down: Not applicable
            LO Down:     0
            LO Up:       0
            Signalling:  [UNCHECKED] AOS [UNCHECKED] LOS
      Ok
    OK
    Module menu
      Radio Control
        Target:    NOAA 15
        T    <- "Tune the radio to this transponder."
        Track
        1. Device: SDRangel-RigCtl
        Engage
```

Gpredict momentarily connects to SDRangel, then looses connection.

Below is the output from SDRangel after clicking "Engage":
```
<click "Engage" in Gpredict>
2023-10-01 19:22:52.632 (D) RigCtlServerWorker::setFrequency: set device frequency 137502626.000000 OK
2023-10-01 19:22:52.632 (D) RTLSDRInput::handleMessage: MsgConfigureRTLSDR
2023-10-01 19:22:52.632 (D) RTLSDRInput::applySettings: force:  false " m_centerFrequency: 137502626"
2023-10-01 19:22:52.632 (D) DeviceSampleStatic::calculateSourceDeviceCenterFrequency:  frequencyShiftScheme:  0  desired center freq:  137502626  Hz  device center freq:  137502626  Hz  device sample rate:  1024000 S/s  Actual sample rate:  64000 S/s  center freq position code:  2  image frequency:  137502626 Hz
2023-10-01 19:22:52.633 (W) RigCtlServerWorker::setFrequency: get channel offset frequency error 404: There is no channel with index -1
2023-10-01 19:22:52.635 (D) RigCtlServerWorker::setFrequency: set device frequency 137502626.000000 OK
2023-10-01 19:22:52.635 (W) RigCtlServerWorker::setFrequency: get channel offset frequency error 404: There is no channel with index -1
2023-10-01 19:22:52.685 (D) RTLSDRInput::applySettings: rtlsdr_set_center_freq(137502626)
2023-10-01 19:22:52.685 (D) RTLSDRInput::handleMessage: MsgConfigureRTLSDR
2023-10-01 19:22:52.685 (D) DSPDeviceSourceEngine::handleInputMessages: message: DSPSignalNotification
2023-10-01 19:22:52.685 (D) RTLSDRInput::applySettings: force:  false " m_centerFrequency: 137502626"
2023-10-01 19:22:52.686 (D) DSPDeviceSourceEngine::handleInputMessages: DSPSignalNotification:  m_sampleRate:  64000  m_centerFrequency:  137502626
2023-10-01 19:22:52.686 (D) DeviceSampleStatic::calculateSourceDeviceCenterFrequency:  frequencyShiftScheme:  0  desired center freq:  137502626  Hz  device center freq:  137502626  Hz  device sample rate:  1024000 S/s  Actual sample rate:  64000 S/s  center freq position code:  2  image frequency:  137502626 Hz
2023-10-01 19:22:52.686 (D) DSPDeviceSourceEngine::handleInputMessages: forward message to  SpectrumVis
2023-10-01 19:22:52.686 (D) DSPDeviceSourceEngine::handleInputMessages: DSPSignalNotification: guiMessageQueue: 0x560cf4f17158
2023-10-01 19:22:52.738 (D) RTLSDRInput::applySettings: rtlsdr_set_center_freq(137502626)
2023-10-01 19:22:52.739 (D) RTLSDRGui::handleInputMessages: message: RTLSDRInput::MsgConfigureRTLSDR
2023-10-01 19:22:52.739 (D) DSPDeviceSourceEngine::handleInputMessages: message: DSPSignalNotification
2023-10-01 19:22:52.739 (D) DSPDeviceSourceEngine::handleInputMessages: DSPSignalNotification:  m_sampleRate:  64000  m_centerFrequency:  137502626
2023-10-01 19:22:52.739 (D) DSPDeviceSourceEngine::handleInputMessages: forward message to  SpectrumVis
2023-10-01 19:22:52.739 (D) DSPDeviceSourceEngine::handleInputMessages: DSPSignalNotification: guiMessageQueue: 0x560cf4f17158
2023-10-01 19:22:52.740 (D) RTLSDRGui::updateFrequencyLimits: delta: 0 min: 24000 max: 2400000
2023-10-01 19:22:52.741 (D) RTLSDRGui::handleInputMessages: message: RTLSDRInput::MsgConfigureRTLSDR
2023-10-01 19:22:52.741 (D) RTLSDRGui::updateFrequencyLimits: delta: 0 min: 24000 max: 2400000
2023-10-01 19:22:52.742 (D) RTLSDRGui::handleInputMessages: message: DSPSignalNotification
2023-10-01 19:22:52.742 (D) RTLSDRGui::handleInputMessages: DSPSignalNotification: SampleRate:64000, CenterFrequency:137502626
2023-10-01 19:22:52.742 (D) GLSpectrumGUI::handleInputMessages: message: GLSpectrumView::MsgReportSampleRate
2023-10-01 19:22:52.743 (D) RTLSDRGui::handleInputMessages: message: DSPSignalNotification
2023-10-01 19:22:52.743 (D) RTLSDRGui::handleInputMessages: DSPSignalNotification: SampleRate:64000, CenterFrequency:137502626
2023-10-01 19:22:52.743 (D) GLSpectrumGUI::handleInputMessages: message: GLSpectrumView::MsgReportSampleRate
2023-10-01 19:22:52.743 (D) SpectrumVis::handleMessage: DSPSignalNotification:  centerFrequency:  137502626  sampleRate:  64000
2023-10-01 19:22:52.743 (D) SpectrumVis::handleMessage: DSPSignalNotification:  centerFrequency:  137502626  sampleRate:  64000
2023-10-01 19:22:53.615 (W) RigCtlServerWorker::setFrequency: get channel offset frequency error 404: There is no channel with index -1
2023-10-01 19:22:54.617 (W) RigCtlServerWorker::setFrequency: get channel offset frequency error 404: There is no channel with index -1
2023-10-01 19:22:55.617 (W) RigCtlServerWorker::setFrequency: get channel offset frequency error 404: There is no channel with index -1
```
This output indicates that there is likely an issue with the SDRangel "channel" code.

If I've done something wrong, please let me know.
If not, then please fix this problem.
Thank you.

f4exb, 2023-10-02:
I guess you should add at least one channel. If you have none it is normal that the list of channels remains empty and therefore there is no target for the rigctl feature.

gvimlag, 2023-10-02:
My mistake.
Thank you for pointing it out.

closed 2023-11-17 by github-actions[bot]