f4exb/sdrangel #1847: Ubuntu 22.10 SDRangel install via Apps store.

agentpatience, 2023-10-14:
Hello, I'm new to this but I get an error when I try to install SDR Angel on Ubuntu 22.10 saying:

/usr/bin$ sdrangel
2023-10-14 18:31:03.771 (F) This application failed to start because it could not find or load the Qt platform plugin "wayland-egl"
in "".

Available platform plugins are: eglfs, linuxfb, minimal, minimalegl, offscreen, vnc, xcb.

Reinstalling the application may fix this problem.
Aborted (core dumped)

What command fixes this?

f4exb, 2023-10-15:
Try to install the .deb file you will find in the releases section here. It is for 22.04 but may work for 22.10. The best option would still be to compile from source.

agentpatience, 2023-10-15:
I cant find the deb file? Also, would you give me instructions to build from source. I dont know the process? Thanks!

srcejon, 2023-10-15:
https://github.com/f4exb/sdrangel/releases

https://github.com/f4exb/sdrangel/wiki/Compile-from-source-in-Linux

neural-loop, 2023-11-06:
Is there supposed to be a .deb version here? 
[image]

srcejon, 2023-11-06:
> Is there supposed to be a .deb version here? [image]

In master.tar.gz.

closed 2023-12-23 by github-actions[bot]