f4exb/sdrangel #1850: DSD Demodulator not decoding

c3kkos, 2023-10-17:
Tried both windows and linux latest versions

I tried to decode some DMR from a sdrangel server with no luck.

That made me download a .wav file from various site, convert them in WAVs 16bit PCM STEREO as sdrangel's docs saying, and feeding them into sdrangel via the "FileInput" source.

everything seems to be good to go but.....

DSD Demodulator is doing NOTHING. I see from the power bar that its recieving data (sound) but nothing happens.

No logs, no constellation... what i'm doing wrong?

I will attach a screenshot of the version, and the file used as sample (i've used plenty.... DMR, D-STAR.... no luck with any at all!)

[image]
[image]

srcejon, 2023-10-17:
If I recall correctly, the DSD Demod doesn't work well if sample rate is below 48kHz, which appears to be the case with your .wav file.

Also, are you sure those .wav files are IQ baseband recordings?

srcejon, 2023-10-17:
That doesn't appear to be an IQ file. The left and right channels appear identical.

c3kkos, 2023-10-17:
Done another test with linux version. In linux i can use pipewire to pipe inside sdrangel the audio from firefox browser. Then, go to DMR SAMPLES pages and inject the sound. SOMETHING happens. So: windows build do not allow any audio input devices, when we would need at least the local stereo loopback.

PCM is not i/q data (of course....) i thought, since we got a fileinput, dsd would work with sound, rather then raw i/q samples.. and indeed it (kinda) work...

srcejon, 2023-10-17:
What web site are you referring to?

In Windows, you can see the constellation for the file you uploaded, if you adjust the input frequency offset a little (E.g. delta f to 50 Hz)The problem appears to be it isn't an IQ recording.

f4exb, 2023-10-17:
I confirm you need an input sample rate of at least 48 kS/s. The DSD demod uses several samples per symbol to be able to decode correctly.

c3kkos, 2023-10-18:
Ok, done 1 more test with proper .wav file provided by:

https://sdrplay.com/resources/IQ/dmr.zip

Now the plugin "reads" but cannot render realtime voice, just a few "glitches"

[image]

c3kkos, 2023-10-18:
OK Done proper test with proper .wav file provided by:

https://sdrplay.com/resources/IQ/dmr.zip

and i get proper voice, so dsd plugin is working.

I still cant see constellation but maybe this is due to 4k monitor. (too thin points)

Next step is decoding something from real antenna and not a recording. more tests feedbacks asap

srcejon, 2023-10-18:
I can see the constellation on a 4k monitor.

c3kkos, 2023-10-19:
OK done some tests.

First of all, sdrangel quits always with a segmentation fault. This is not a problem, but sometimes it goes segfault even only loading the dsd demodulator.

Sometimes, at random, dsd demod simply do not work. You can set squelch -100 and still icon not goin green

In "normal" usage, dsd mod gives you weird sound only using thermal noise (example...)

I found a workaround to make it work 100% times:

start with audio source. load dsd onto that device

then load the RTL (usb connected) only after that switch device on dsd mod, from audio to rtl sdr.

Weird enough!

Ah! also have some issue with pipewire. I switched back to pulse output, it seems more stable.

I'm running arch linux full updated

f4exb, 2023-10-20:
> Ah! also have some issue with pipewire. I switched back to pulse output, it seems more stable.

No wonder... I remember someone willing to push usage of pipewire as the next World's marvel. Well it is probably not...

c3kkos, 2023-11-02:
This will go off topic, but let me say i've been using Pipewire and Wireplumber as session manager and IMHO is a VERY good piece of software! Piping audio made easy and reliable. Of course it need knowledge to work with it

closed 2023-12-18 by github-actions[bot]