f4exb/sdrangel #1851: Adding support for WOLF DDC

autodiag, 2023-10-21:
I read somewhere, apparently directly on the author's website, about the WOLF DDC support plan
is this support still in play or has this idea been abandoned

I think it would be a very interesting connection, plus the author is quite proficient at adding features, fixes and support.

f4exb, 2023-10-24:
The Wolf that I start to know quite well having worked out its guts to fix the Chinese version mistakes (!) is controlled via CAT (TS-2000 type) and has an audio in and audio out for I/Q transfer. All with the same USB cable. Hence it is supported by this plugin: https://github.com/f4exb/sdrangel/tree/master/plugins/samplemimo/audiocatsiso Which is a generic plugin for such setups. 

It is to my knowledge the only rig of this type (like a "normal" radio) giving access to random I/Q modulation input. Limited to a bandwidth of 48 kHz but this is still room for a lot of experimentations.

autodiag, 2023-10-25:
yes, the Chinese version, like almost everything else they hastily produce, needs a lot of repairs..I had to put another wifi module on mine, completely re-fit the filters with quality components, including RF switches that are unsuitable, antenna switching didn't work, LNA didn't work, etc...etc....and the processing and settling took its toll :D ...
I ask because it would be nice to use some decoders that SDRAngel has in conjunction with Wolf..I tried to use it as a sound card for example for DMR but it doesn't work

f4exb, 2023-10-25:
AudioCATSISO needs Hamlib precisely for CAT. I am not sure this works in all environments.

autodiag, 2023-10-27:
Is there a website where these errors are described ?I have a few more things I don't like

f4exb, 2023-10-28:
Well... I could not find a synthetic page summarizing these updates. There is a channel in Telegram specifically dedicated to the Chinese clones: https://t.me/ZT7500 There is also the "main" channel which link is found in UA3REO's repository but people there are not very nice to Chinese "cloners"... Posts are in Russian but you can translate in Telegram and use Google translate to post. Some people also post in English.

autodiag, 2023-10-28:
thanks..i known too

closed 2023-12-13 by github-actions[bot]

clzls, 2024-06-20:
> Hence it is supported by this plugin: https://github.com/f4exb/sdrangel/tree/master/plugins/samplemimo/audiocatsiso Which is a generic plugin for such setups.

I would like to know how I can get this plugin ready on my SDRangel installation using prebuilt Windows/macOS installers... Or we do have to face complex building process? :(