f4exb/sdrangel #1859: Swagger code gen incorrect for arrays of integer

srcejon, 2023-10-24:
It seems the swagger code generation for arrays of integers is incorrect.

As an example (but there are others), in VORLocalizer.yaml, we have:

    columnIndexes:
      type: array
      items:
        type: integer

When getting feature settings through the Swagger webpage, the array returned is always empty:

    "columnIndexes": [],

This can be fixed by changing the generated code SWGVorLocalizerSettings.cpp from:

        toJsonArray((QList<void*>*)column_indexes, obj, "columnIndexes", "");

To:

        toJsonArray((QList<void*>*)column_indexes, obj, "columnIndexes", "qint32");

But this fix obviously needs to be done in the generator.

(Arrays of strings work ok - they have the last parameter set to "QString")

I notice many other examples of arrays in the yaml files actually uses arrays of structures with a single number/integer element - was that a workaround to this, perhaps? That works, but obviously a little less efficient.

srcejon, 2023-10-24:
Looking at the code gen, it sort of looks like this template is used to generate the code:

https://github.com/swagger-api/swagger-codegen/blob/master/modules/swagger-codegen/src/main/resources/qt5cpp/model-body.mustache

But the code isn't exactly the same, so I'm not quite sure.

That template has:

     toJsonArray((QList<void*>*){{name}}, obj, "{{baseName}}", "{{complexType}}");

Which suggests the complexType parameter isn't being set to qint32.

srcejon, 2023-10-25:
Looks like an upstream bug (https://github.com/swagger-api) with swagger-codegen, as I see the same problem using that.

f4exb, 2026-01-04:
The code being used comes from an old fork of mine and actually can be seen here: https://github.com/f4exb/swagger-codegen/blob/sdrangel/modules/swagger-codegen/src/main/resources/qt5cpp/model-body.mustache#L113 
But I am not proficient in the mustache (sort of) language