f4exb/sdrangel #1860: Error in compiling Soapy SDR

auskimus, 2023-10-26:
Hi, having difficulty locating what went wrong compiling SoapySDR
 $ make -j $(nproc) install
on Ubuntu22.04
[image]

closed 2023-12-12 by github-actions[bot]

srcejon, 2024-02-26:
Problem appears to be python related. Make sure you have python3 installed as per the instructions.

Then have a look at what cmake outputs - I get:

-- Could NOT find Python3InterpDbg (missing: PYTHON3_DBG_EXECUTABLE) 
-- PYTHON3INTERP_FOUND: TRUE
-- PYTHON3_EXECUTABLE: /usr/bin/python3
/opt/build/SoapySDR/python3/../python/get_python_lib.py:4: DeprecationWarning: The distutils package is deprecated and slated for removal in Python 3.12. Use setuptools or check PEP 632 for potential alternatives
  from distutils.sysconfig import get_python_lib
/opt/build/SoapySDR/python3/../python/get_python_lib.py:4: DeprecationWarning: The distutils.sysconfig module is deprecated, use sysconfig instead
  from distutils.sysconfig import get_python_lib
-- PYTHON3_INSTALL_DIR: ${prefix}/lib/python3.10/site-packages
-- Found Python3Libs: -L/usr/lib/python3.10/config-3.10-x86_64-linux-gnu -L/usr/lib/x86_64-linux-gnu  -lcrypt -ldl  -lm -lm  
-- PYTHON3LIBS_FOUND: TRUE
-- PYTHON3_INCLUDE_DIRS: /usr/include/python3.10;/usr/include/python3.10
-- PYTHON3_LIBRARIES: -L/usr/lib/python3.10/config-3.10-x86_64-linux-gnu -L/usr/lib/x86_64-linux-gnu  -lcrypt -ldl  -lm -lm

auskimus, 2024-02-26:
> @auskimus<https://github.com/auskimus> HI,I met the same problem. Have you solved the problem

Hi

The guy seems to not answering questions about some issues already mentioned in the Read me
I just installed a snaped Version.

Regards