f4exb/sdrangel #1862: Crashes on RTTY

Rippey5742, 2023-10-27:
I cant use RTTY, I want to use RTTY. This is my favorite modulation. 

Every time I go to try and use it, it will crash the app or freeze the program. 
[rtty crash.txt](https://github.com/f4exb/sdrangel/files/13184208/rtty.crash.txt)

Can I get a hand on this issue?

srcejon, 2023-10-27:
This looks a bit suspicious: 

2023-10-26 21:39:18.191 (D) RttyModSource::applyChannelSettings:  channelSampleRate:  0  channelFrequencyOffset:  0  rfBandwidth:  340
2023-10-26 21:39:18.191 (D) m_samplesPerSymbol:  0  ( 0 / 45.45 )

srcejon, 2023-10-27:
More likely to be a problem with RaisedCosine filter normalisation. 

RTTYMod runs at baseband sample rate, which is 2.4M for you, and because of the high sample rate, it's trying to create a filter with a massive number of taps, and then normalization takes forever.

Rippey5742, 2023-10-27:
[rtty testing.log](https://github.com/f4exb/sdrangel/files/13193508/rtty.testing.log)
Lowest sample rate I can go 1 million. 

Based off the thought it was going to take a large amount of time to its thing, i left it sit. I was able to finally get it to transmit, it just took an extended time. Here is the log. It had a few errors at the end then it operated. I seen you created that other post, but wasn't sure on what one to respond back in. 

Took around 19 minutes.

srcejon, 2023-10-27:
The above patch should help, but will need also need to rewrite another bit of code as well.