f4exb/sdrangel #1863: Loading select plugins at program start

hb9dug, 2023-10-27:
Not an issue but a question

It's too complicated to use!   That's the answer I often get when I suggest using sdrangel for DATV. To provide a partial answer, I'd like to limit the plugins loaded at startup to those required for DATV operation.I  manage to do this by deleting the dll in the plugins directory, but  it's not very graceful... Especially since each time I update, I have to  repeat the operation.  :-(Question:Is it possible to create a configuration file containing the list of plugins to be loaded and then, for example: sdrangel -L load_plugins.conf to load them?Merci for your help,Michel

srcejon, 2023-10-27:
A command line option makes it more complicated, not less, surely? And what does it really simplify? It just removes a few options from some combo boxes.

If you want to make it easier for people to use for a specific application, why not provide a configuration file that includes all needed device, spectrum and channel settings, so they just have to load that configuration? (E.g. star button on toolbar)

Or, if you're scripting something up for people, then perhaps we could add a command line option that specifies a configuration to load on startup.

Being able to disable plugins might be useful to help speedup startup time though. For that, I would suggest they be enabled via a checkbox in the Help > Loaded Plugins dialog, rather than an additional config file.

f4exb, 2023-10-28:
I agree that limiting the plugins being loaded will only have an effect on startup time not complexity which I think is a non issue since you can load a configuration with the "star" button on the top bar. Configurations are serializable and can be shared among users. This is not completely trivial but is done only once. Then if you exit with this configuration it will be loaded automatically at the next program start.

However if you like scripting there is also the possibility to configure a complete setup with https://github.com/f4exb/sdrangel/blob/master/scriptsapi/config.py and a JSON file.

closed 2023-12-13 by github-actions[bot]