f4exb/sdrangel #1866: End of Train (EOT, 457.9375 MHz) packets decoder

abelouso, 2023-10-28:
Hello,
 Would it be possible to add the End of Train packets decoder on 457.9375 MHz? It is a relatively simple known frequency modulated format.

Thank you for reading.

abelouso, 2023-12-06:
Reference implementation can be found in:

https://github.com/russinnes/EOTDecode/

https://github.com/ereuter/PyEOT

srcejon, 2023-12-06:
1200k FSK.. can probably use the Packet demod.

Can you capture some IQ recordings?

abelouso, 2023-12-31:
Recorded I/Q samples (raw) at 457.937500 MHz, 240000 sample rate with decimation of 2: 120.0 ksps
[gqrx_20231231_191106_457937500_120000_fc.raw.gz](https://github.com/f4exb/sdrangel/files/13802052/gqrx_20231231_191106_457937500_120000_fc.raw.gz)

srcejon, 2024-03-02:
Thanks. Was that train stationary?

If you can get another IQ recording of a train braking, it would be useful, thanks.

f4exb, 2024-03-09:
DEployed in v7.19.0

abelouso, 2024-04-01:
Not sure if this could be re-opened or a new enhancement request should be created. 
In the HOT/EOT table there should be an option to update row for an already existing IDs rather than always append received packets as new rows. That would allow to monitor trains by sorting the table columns based on ID, and/or time. Also, adding a signal strength for each received packet as the column in the table would help with approximating the train's proximity and/or direction of travel (away or towards).