f4exb/sdrangel #1867: new plugins RTTY and PSK31

it2024swl, 2023-10-29:
hello, despite I have unistalled the old version before installing the new one, there's no way to see the RTTY decoder with new interface and the PSK31 channel.
In the "Add channel" list the PSK31 is simply not listed (but the modpsk31.dll is in the directory of plugins), and launching the RTTY decoder, it still appears with the old interface.

In Help menu, "loaded plugins", the PSK31Modulator ver. 7.17.0 is there and I have 2 RTTY: RTTY modulator Ver. 7.11.0 and an RTTY demodulator ver 7.17.0

[image]

what am I doing wrong?

thanks
Oscar

srcejon, 2023-10-29:
Please can you attach your log file?

srcejon, 2023-10-29:
Just to clarify, there is no PSK31 Demodulator.

You'll only see the PSK31 Modulator if you have a TX device open.

RTTY Demodulator will be the latest, despite the version number. I just forgot to update it.

it2024swl, 2023-10-29:
AH OK, sorry I thought the PSK31 was a demodulator. so no worries about that I've no transmitter connected. For the RTTY please can you update it?

thank
Oscar

f4exb, 2023-10-30:
This is only for documentary purposes and won't change the fact that the changes are here. In fact it is my fault since I ama responsible for the releasing of new versions. I just missed that one. In fact before updating the plugins versions I should do `git diff --name-only v7.16.0 | sort` here v7.16.0 is the tag of the preceding release. I see others also probably missing: demodnavtex, amdemod, noisefigure, radioclock.

closed 2023-12-01 by srcejon