f4exb/sdrangel #1868: 3 D map disappears

choupette-change, 2023-10-29:
After several minutes of operation, the 3 D map disappears ???

[sdrangel.log](https://github.com/f4exb/sdrangel/files/13198838/sdrangel.log)
[image]

srcejon, 2023-10-29:
The log doesn't show anything.

What graphics card do you have? How much memory does it have? Some of the AIS models are quite large.

Do you have the same problem when not running the AIS demods?

choupette-change, 2023-10-30:
During my first problem, I was in a virtual machine under Windows 11. Now, I installed Ubuntu 22.04 directly on my PC and there no problem of disappearance but the display of the 3 D map is much slower than the 2D...
My current situation (see screenshot below)

[image]

closed 2023-12-15 by github-actions[bot]