f4exb/sdrangel #1873: Frequency Tracker for SSB fine tuning

brcros, 2023-11-01:
Hello,
Using the cool Frequency Scanner, i noticed that delta frequency values can be really differents when jumps are done, and so, decided to use the Frequency Tracker to correct the SSB demodulator  delta frequency in order to listen a real human voice ( prevent duck voice or robot voice). 
I launched the freqtracking.py script that seemsn to work correctly, as it changeq the delta frequency when tracker decide to send a value.
Well, what i don't understand is when the tracker is supposed to apply the correction. Actually, it seems that when the gap is less than 100 Hz, the correction is not applied/sent. And most of the time, it's around 50 Hz needed to jump to the good carrier. I'm quiet sure of that, because when i decrease the delta frequency on the tracker of 100 Hz, the correction is sent, and magic, well calculated. This works.
I'm using FFL, with 8 on decimation with a USB 4kHz bandwith on air band... a goog practice.
May be there is an internal settings for the minimal gap to send, else, a little knob would be perfect. 
Another way, shoul b a timer ( with a knob) to apply the correction even very small...
The 2 knobs is may be too much, but why not. Frequency tracker is a great tool. Thank you.

f4exb, 2023-11-02:
I am surprised that this would work for SSB anyway. The Frequency Tracker is meant to follow a carrier and a SSB signal has not so it will jump all over the place trying to tune to a sort of average of the maximum frequency of the voice signal. See the video here:

https://github.com/f4exb/sdrangel/assets/6192319/c6cf92ca-36b4-4c89-bda1-d27b076ca913

It is possible to use the Frequency Scanner for fixed frequency SSB channels like aeronautical frequencies on HF. Even if at the moment it is meant to track symmetric signals around a carrier it can still sense one sideband or the other and stop on a transmission.

brcros, 2023-11-02:
I missed this point because i only played with the Frequency Scanner on aeronautical frequencies, that's it. So , i had a carrier. And often, i can hear this carrier if the emiting is too shifted on the right. That's precisely why i tried the frequency scanner to catch the carrier and put it in the low cutoff part :-) My bad, the title of the issue should be "Frequency Tracker for aeronautical fine tuning"
As i said, it works using FFL, on air band, even if signal is not symetrical. I guess the carrier is well detected even if there's nothing on left. The great thing would be to trig shift even for low value , such 30 Hz. A knob to change it would be perfect. 
( or

closed 2023-11-02 by brcros