f4exb/sdrangel #1875: SDRPlay API v3 init error

c3kkos, 2023-11-02:
RSP1 **fail with v3 api** to init with this error:

**_sdrplay_apiService[1388]: [1580]: devIdx0: sdrplay_apiService_rsp: Init: ERROR: Tuner1: initHw returned error 7_**

I'm using the api service:

systemctl status sdrplay.service 
● sdrplay.service - SDRplay API Service
     Loaded: loaded (/etc/systemd/system/sdrplay.service; disabled; preset: disabled)
     Active: active (running) since Thu 2023-11-02 10:48:19 CET; 59s ago

with latest sdrangel:

sdrangel -v
SDRangel 7.17.0

on full updated arch linux.

~ uname -r
6.5.9-zen2-1-zen

WITH sdrplay service ON, sdrangel correctly list SDRPLAY v3 api

WITHOUT sdrplay service, rsp1 can still be selected and used without problems

SDRPLAY aurs installed:

1 aur/libsdrplay 3.07.1-1 [+9 ~0.00] [Installed]
    Modules for the SDRplay receiver
8 aur/soapysdrplay3-git r63.f0a8439-1 [+1 ~0.03] [Installed: r77.534a173-1]
    Soapy SDR module for SDRPlay

srcejon, 2023-11-02:
That's probably a question for SDRplay support.

c3kkos, 2023-11-03:
Version 3, the latest avalaible. Working good with soapy and other sw

srcejon, 2023-11-05:
What's the full version number?

And can you please attached your SDRangel log file? Thanks

c3kkos, 2023-11-08:
yes.

i got this:

https://aur.archlinux.org/packages/libsdrplay

and some1 has the same problem:

> simona commented on [2023-09-04 16:54 (UTC)](https://aur.archlinux.org/packages/libsdrplay#comment-932341)
> launching SDRconnect on arch linux it write "Device initialisation failed. Error = 1 (sdrplay_api_Fail)"

(its on the bottom of that page)

srcejon, 2023-11-08:
That's not the same error message.

But I think you need to ask SDRplay support: https://www.sdrplay.com/supportmenu/ - the error message doesn't appear to be from SDRangel.

closed 2023-11-08 by srcejon