f4exb/sdrangel #1877: Code signing on Windows

srcejon, 2023-11-02:
Download & install on Windows would be a great deal more user friendly, if the installer was signed (to bypass the maze of dialog boxes Windows throws up encouraging you not to run unsigned code).

Normally, this has a cost of several hundred $ a year for the certificates, but I came across [signpath.io](https://about.signpath.io/product/open-source) who appear to offer a free signing service & shared certificate for open-source software - and having tried a couple of apps signed with it, it appears to help (the installers run straight away).

It seems they need the build process to be verifiable - and currently support appveyor - with github actions in preview version.

Are you happy if I get in touch, to see if they will support us?

f4exb, 2023-11-03:
Hello @srcejon,

well if it does not make the CI more complex and difficult to maintain why not? ...

Brgds.

CRD716, 2024-09-09:
Was there ever any progress on this?

srcejon, 2024-09-23:
> Was there ever any progress on this?

No, as was waiting for Github Actions support to be available. This has just been released so will take a look.

PR #2344 (2024-12-02): Windows code signing update
Remove unzip step, as it looks like it's not needed. And increase timeout to 1hour.

Hopefully that should now work and we can close #1877
closed 2024-12-02 by f4exb