f4exb/sdrangel #1878: Android application crashes whenever there is a sync frame detected in DSD

Auxilus, 2023-11-04:
```
22:34:20.306  A  Fatal signal 6 (SIGABRT), code -1 (SI_QUEUE) in tid 2710 (QThread), pid 32540 (org.sdrangel)
22:34:20.420  A  pid: 32540, tid: 2710, name: QThread  >>> org.sdrangel <<<
22:34:20.428  A        #02 pc 00000000000762ac  /data/app/~~lZ1KycWVgZe8nV60ZVd9-w==/org.sdrangel-lgKPiyTalM2eWH5_xSkBxQ==/lib/arm64/libdsdcc.so (DSDcc::DSDDMR::processSlotTypePDU()+508) (BuildId: b456994073042a14d3751b23a2f83c99c4098f36)
22:34:20.429  A        #03 pc 0000000000074c38  /data/app/~~lZ1KycWVgZe8nV60ZVd9-w==/org.sdrangel-lgKPiyTalM2eWH5_xSkBxQ==/lib/arm64/libdsdcc.so (DSDcc::DSDDMR::processData()+132) (BuildId: b456994073042a14d3751b23a2f83c99c4098f36)
22:34:20.429  A        #04 pc 000000000002e5b4  /data/app/~~lZ1KycWVgZe8nV60ZVd9-w==/org.sdrangel-lgKPiyTalM2eWH5_xSkBxQ==/lib/arm64/libdemoddsd_arm64-v8a.so (DSDDemodSink::feed(std::__ndk1::__wrap_iter<Sample const*> const&, std::__ndk1::__wrap_iter<Sample const*> const&)+1216) (BuildId: 5a7f8e679d0c95e12faf346aef4c793c083e0293)
22:34:20.429  A        #05 pc 0000000000a15138  /data/app/~~lZ1KycWVgZe8nV60ZVd9-w==/org.sdrangel-lgKPiyTalM2eWH5_xSkBxQ==/lib/arm64/libsdrbase_arm64-v8a.so (DownChannelizer::feed(std::__ndk1::__wrap_iter<Sample const*> const&, std::__ndk1::__wrap_iter<Sample const*> const&)+628) (BuildId: cfc63f21e441d75aede286ccf4dfd69e506b54f2)
22:34:20.429  A        #06 pc 00000000000301fc  /data/app/~~lZ1KycWVgZe8nV60ZVd9-w==/org.sdrangel-lgKPiyTalM2eWH5_xSkBxQ==/lib/arm64/libdemoddsd_arm64-v8a.so (DSDDemodBaseband::handleData()+204) (BuildId: 5a7f8e679d0c95e12faf346aef4c793c083e0293)
22:34:20.429  A        #07 pc 0000000000483e7c  /data/app/~~lZ1KycWVgZe8nV60ZVd9-w==/org.sdrangel-lgKPiyTalM2eWH5_xSkBxQ==/lib/arm64/libQt5Core_arm64-v8a.so (QObject::event(QEvent*)+456) (BuildId: 1f19d00d68ade81c7dfcfcae973f3fadf4d3ef32)
22:34:20.429  A        #08 pc 000000000026efe4  /data/app/~~lZ1KycWVgZe8nV60ZVd9-w==/org.sdrangel-lgKPiyTalM2eWH5_xSkBxQ==/lib/arm64/libQt5Widgets_arm64-v8a.so (QApplicationPrivate::notify_helper(QObject*, QEvent*)+296) (BuildId: 9b49e193620cac8290676ecd3d5378fad8fc5c08)
22:34:20.429  A        #09 pc 0000000000270404  /data/app/~~lZ1KycWVgZe8nV60ZVd9-w==/org.sdrangel-lgKPiyTalM2eWH5_xSkBxQ==/lib/arm64/libQt5Widgets_arm64-v8a.so (QApplication::notify(QObject*, QEvent*)+560) (BuildId: 9b49e193620cac8290676ecd3d5378fad8fc5c08)
22:34:20.429  A        #10 pc 000000000046a2f8  /data/app/~~lZ1KycWVgZe8nV60ZVd9-w==/org.sdrangel-lgKPiyTalM2eWH5_xSkBxQ==/lib/arm64/libQt5Core_arm64-v8a.so (QCoreApplication::notifyInternal2(QObject*, QEvent*)+176) (BuildId: 1f19d00d68ade81c7dfcfcae973f3fadf4d3ef32)
22:34:20.429  A        #11 pc 000000000046afb0  /data/app/~~lZ1KycWVgZe8nV60ZVd9-w==/org.sdrangel-lgKPiyTalM2eWH5_xSkBxQ==/lib/arm64/libQt5Core_arm64-v8a.so (QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*)+432) (BuildId: 1f19d00d68ade81c7dfcfcae973f3fadf4d3ef32)
22:34:20.429  A        #12 pc 00000000004a018c  /data/app/~~lZ1KycWVgZe8nV60ZVd9-w==/org.sdrangel-lgKPiyTalM2eWH5_xSkBxQ==/lib/arm64/libQt5Core_arm64-v8a.so (QEventDispatcherUNIX::processEvents(QFlags<QEventLoop::ProcessEventsFlag>)+76) (BuildId: 1f19d00d68ade81c7dfcfcae973f3fadf4d3ef32)
22:34:20.429  A        #13 pc 0000000000468120  /data/app/~~lZ1KycWVgZe8nV60ZVd9-w==/org.sdrangel-lgKPiyTalM2eWH5_xSkBxQ==/lib/arm64/libQt5Core_arm64-v8a.so (QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>)+356) (BuildId: 1f19d00d68ade81c7dfcfcae973f3fadf4d3ef32)
22:34:20.429  A        #14 pc 0000000000348818  /data/app/~~lZ1KycWVgZe8nV60ZVd9-w==/org.sdrangel-lgKPiyTalM2eWH5_xSkBxQ==/lib/arm64/libQt5Core_arm64-v8a.so (QThread::exec()+96) (BuildId: 1f19d00d68ade81c7dfcfcae973f3fadf4d3ef32)
22:34:20.430  A        #15 pc 0000000000349580  /data/app/~~lZ1KycWVgZe8nV60ZVd9-w==/org.sdrangel-lgKPiyTalM2eWH5_xSkBxQ==/lib/arm64/libQt5Core_arm64-v8a.so (BuildId: 1f19d00d68ade81c7dfcfcae973f3fadf4d3ef32)
22:34:20.870  E  SampleSinkFifo::write: (DSDDemod [0:0]) overflow - dropping 98000 samples
```

srcejon, 2023-11-05:
Thanks. Any chance you can make an IQ recording (.wav using File SInk) as it's not crashing for me with the limited tranmissions near me.

Auxilus, 2023-11-05:
I can, once I get back home. Meanwhile, I have a question, how would someone go about debugging this thing? Is there any documentation on the android version?

srcejon, 2023-11-09:
We don't have a way to debug properly on Android at the moment. 

I will try recompiling dsdcc with the latest Android toolchain though, to see if that makes a difference.

srcejon, 2023-11-10:
@f4exb, I think I can see 2 bugs in dsdcc relating to this. 

In DSDDMR::processSlotTypePDU(), we have:

        if (dataType > DMR_TYPES_COUNT)
        {
            m_dataType = DSDDMRDataReserved;
            memcpy(&m_slotText[4], "RES", 3);
        }
        else
        {
            m_dataType = (DSDDMRDataTYpe) dataType;
            memcpy(&m_slotText[4], m_slotTypeText[dataType], 3);
        }

However, I think this should be:

        if (dataType >= DMR_TYPES_COUNT)

As there's only DMR_TYPES_COUNT elements in m_slotTypeText.

But the bug that's causing the crash is I think in fec.cpp in Golay_20_8::decode(unsigned char *rxBits)

rxBits has 20 elements, however, this line of code sometimes writes beyond this:

                rxBits[m_corr[syndromeI][i]] ^= 1; // flip bit

i.e. I see it doing rxBits[21] ^= 1; which corrupts the stack in the calling frame.

You can use this: https://sdrplay.com/resources/IQ/dmr.zip as a testcase IQ file.

srcejon, 2023-11-10:
The next problem is that there is no audio output. This is because MBE lib wasn't being linked.

In dsdcc/cmake/Modules/FindLibMbe.cmake, we have:

    IF (NOT LIBMBE_LIBRARY)
        FIND_LIBRARY(LIBMBE_LIBRARY NAMES ${LIBMBE_NAMES} PATH)
    ENDIF()

This should just be:

    FIND_LIBRARY(LIBMBE_LIBRARY NAMES ${LIBMBE_NAMES} PATH)

So that CMake correctly understands that -DLIBMBE_LIBRARY= specifies a file path rather than library name, and converts it as needed for the Android toolchain.

Auxilus, 2023-11-10:
> The next problem is that there is no audio output. This is because MBE lib wasn't being linked.
>
> In dsdcc/cmake/Modules/FindLibMbe.cmake, we have:
>
> ```
> IF (NOT LIBMBE_LIBRARY) [...]

yes, I forgot to mention the audio issue. If I want to build the app myself, is there any documentation on the toolchain  that I can refer?

srcejon, 2023-11-10:
>  If I want to build the app myself, is there any documentation on the toolchain that I can refer?

No - it's very difficult to build at the moment, as many of the dependencies need patches. I'm trying to submit those upstream, but will take some more time.

Auxilus, 2023-11-16:
> @f4exb, I think I can see 2 bugs in dsdcc relating to this.
>
> In DSDDMR::processSlotTypePDU(), we have:
>
> ```
>     if (dataType > DMR_TYPES_COUNT)
>     {
>         m_dataType = DSDDMRDataReserved; [...]

@srcejon I recompiled dsdcc with these changes and rebuilt the apk. It definitely fixed the crash for me. Audio output is also working after linking libmbe.

srcejon, 2023-11-29:
Have a PR with a fix for this: https://github.com/f4exb/dsdcc/pull/61

closed 2023-11-29 by f4exb