f4exb/sdrangel #1895: TX interpolation >= 16x problem

srcejon, 2023-11-17:
As was discussed on #1892, the M17 modulator did not appear to work with low baseband sample rate (<100k).

I suspect the problem is actually with interpolation when set to 16x or greater.

Here's a picture of TX and RX spectrum of the packet mod, with Int = 8 - which looks OK.

[image]

And here it is with Int = 16

[image]

Which looks like it has a problem.

f4exb, 2023-11-25:
"MyB210" suggests you are using a USRP here. I have tried with a HackRF Tx + RTL-SDR Rx combo and I do not notice the issue:

https://github.com/f4exb/sdrangel/assets/6192319/6a074c5d-d7ab-48f6-bb5c-7510129091fe

Also what modulator do you use? I have used the packet modulator in a loop transmission to obtain similar pulses.

srcejon, 2023-11-25:
Yep, a USRP with with Packet Mod.

In usrpoutputthread.h

       Interpolators<qint16, SDR_TX_SAMP_SZ, 12> m_interpolators;

In hackrfoutputthread:

      Interpolators<qint8, SDR_TX_SAMP_SZ, 8> m_interpolators;

So different data-types / scaling may explain it.

srcejon, 2023-11-25:
BladeRF uses the same interpolators, but I don't have one.

Also, might need something with higher res ADC than RTL SDR to see the problem.

f4exb, 2023-11-25:
Yes I just tried with the BladeRF 2 at an interpolation of 16 and using the Aaronia RTSA as the receiver which cannot suspected of bad dynamic range. Even with a dynamic of 60dB on the spectrum no arifacts are visible.

[image]

srcejon, 2023-11-25:
Ok, thanks. Will try to narrow it down some more here.

sp5wwp, 2024-01-31:
Are there any updates to this?

srcejon, 2024-01-31:
Working on something else at the moment, I'm afraid.

sp5wwp, 2024-09-26:
Is this bug still present?

srcejon, 2024-09-27:
I suspect so, unfortunately.