f4exb/sdrangel #1899: NFM Mod Freq Deviation

srcejon, 2023-11-20:
I think the NFM Mod has the same problem with Frequency Deviation as the M17 Mod - i.e. a missing factor of PI.

    m_modPhasor += (m_settings.m_fmDeviation / (float) m_audioSampleRate) * t1;

Should be:

    m_modPhasor += (M_PI * m_settings.m_fmDeviation / (float) m_audioSampleRate) * t1;

With that change, I would expect with a full-scale audio signal (i.e. setting t0=1), to get peak deviation. However, there's an extra factor of 1.2:

    if (m_settings.m_ctcssOn) {
        t1 = (0.85f * m_bandpass.filter(t) + 0.15f * 0.625f * m_ctcssNco.next()) * 1.2f;
    } else if (m_settings.m_dcsOn) {
        t1 = (0.9f * m_bandpass.filter(t) + 0.1f * 0.625f * m_dcsMod.next()) * 1.2f;
    } else if (m_settings.m_bpfOn) {
        t1 = m_bandpass.filter(t) * 1.2f;
    } else {
        t1 = m_lowpass.filter(t) * 1.2f;
    }

Any idea what that's for?

I suspect the last case should just be:

        t1 = m_lowpass.filter(t);

f4exb, 2023-11-20:
I think this was some empirical adjustment maybe to compensate filter attenuation. There is also a 0.625 factor applied to the CTCSS and DCS modulations supposedly also empirical.

srcejon, 2023-11-25:
> I think this was some empirical adjustment maybe to compensate filter attenuation. 

Here's freq response of the current low pass filter, which doesn't look like it needs it, as 0dB:

[image]

And the bandpass filter looks to be 0dB in passpand too:

[image]

f4exb, 2023-11-26:
Verified how it is for WFM modulator and indeed there is a multiplication by &pi; 
https://github.com/f4exb/sdrangel/blob/master/plugins/channeltx/modwfm/wfmmodsource.cpp#L122

commit 003a04ed (2023-11-26): NFM modulator: applied suggested fixes. Fixes #1899
closed 2023-11-26 by f4exb