f4exb/sdrangel #1901: Channel analyzer: channel scope trigger

keukenrol, 2023-11-23:
Hi

I am trying to get the triggers working in the channel scope, but it does not seem to function as expected.
I have set trace length to a minimum to detect a single pulse. With a sample rate of 20MSPS, this gives me a trace length of 240 microseconds.
I would like to use the one shot trigger. The level is known, thus it is set using the Trig0 trigger level slider. Tried both real, imaginary and magnitude variants. While the signal does show up in freerunning mode, it never seems to actually trigger.

Am I missing something?

Thanks!

srcejon, 2023-11-23:
Can you add a screenshot please?

keukenrol, 2023-11-23:
> Can you add a screenshot please?

I suppose it would be easier if I make a video of it, but here is a screenshot with a summary:
[image]
In freerunning mode (thus, trigger not active), it is clear that the signal level rises above the set trigger level. The green trigger line is below the pulse amplitude. The trigger level is set very low, just above the noise level to make sure it triggers as soon as possible.

[image]
In oneshot, the image remains frozen like this. I would expect to be able to trigger on a pulse as soon as it rises above the set trigger level, and freeze the display.

I also noticed that the framerate of the channel scope is only a few FPS. While the channel spectrum easily runs on 50FPS.

srcejon, 2023-11-23:
For the waveform I'm looking at, it triggers in one-shot mode, but not where I'd expect it to:

[image]

With rising edge trigger, I'd expect it to trigger where the signal crosses upwards across the trigger level, and for that to be time 0.

A smaller issue I see, is that when adjusting the trigger level (when not in single-shot mode), the trigger will occur for each adjustment, even if the adjustment to the level remains outside of the range of the signal.

keukenrol, 2023-11-23:
Do you have the possibility to try on a high sample rate? I have tried it with the RTL-SDR and the HackRF (shown in screenshots).

srcejon, 2023-11-24:
With the above patch, triggering seems to be improved.

However, there seems to be another issue when the sample rate is increased. If I use test source to feed in a square wave or saw wave, once the sample rate is greater than ~90kSa/s, the trace appears corrupted.

[image]

[image]

keukenrol, 2023-11-24:
> With the above patch, triggering seems to be improved.
> 
> However, there seems to be another issue when the sample rate is increased. If I use test source to feed in a square wave or saw wave, once the sample rate is greater than ~90kSa/s, the trace appears corrupted.
> 
> [image]
> 
> [image]
> 
> 

Thanks for checking it out. If you need me to do some tests, let me know. I do have a hackrf and 2 rtl-sdr's at my disposal.

I am trying to visualise an AM modulated pulse of only a few microseconds. If the channel scope is corrupted it is difficult to visualise the true waveform. Any idea wether the iq stream also contains errors, or is it linked directly to the channel scope?

Is there a possibility to play back an IQ stream in blocks? (for example blocks of 20K samples) That way one can accurately check the incoming signal. Like you would play back a video frame by frame.

Thanks for your efforts!

srcejon, 2023-11-24:
The IQ stream is fine - I recorded it using the File Sink then viewed in Audacity. You can use that to edit IQ files and cut them up.

keukenrol, 2023-11-24:
Yeah, but it seems audacity only supports sample rates of up to 384ks/s. I made a short wav file of my 20Ms/s stream. While I was able to visualise it, cutting and inserting it in a new project or track does not work. But issues of audacity are out of scope here of course.

f4exb, 2023-11-25:
> However, there seems to be another issue when the sample rate is increased. If I use test source to feed in a square wave or saw wave, once the sample rate is greater than ~90kSa/s, the trace appears corrupted.

I suppose you mean something like this:

https://github.com/f4exb/sdrangel/assets/6192319/41b17602-76ed-461f-a352-4df45dce43ac

The artifacts in the device spectrum appear to be regular which is not the case in the channel analyzer spectrum. I guess this seems to indicate that the issue is in the channel analyzer not the test source.

srcejon, 2023-11-25:
> I suppose you mean something like this:

Yep

> The artifacts in the device spectrum appear to be regular which is not the case in the channel analyzer spectrum. I guess this seems to indicate that the issue is in the channel analyzer not the test source.

Yeah - I used the File Sink to record the Test Source output and it looked OK in Audacity.

keukenrol, 2023-12-07:
On version 7.17.2 the trigger seems to function a lot better already. Will test in the coming days a bit more. Thanks!

srcejon, 2026-09-29:
Should be fixed

closed 2026-09-29 by srcejon