f4exb/sdrangel #1912: Device, Channel and Feature Ids

srcejon, 2023-12-05:
There are a number of plugins that use Device "T0", Channel "R2:3" and Feature "F0:2" identifiers, to allow a user to specify other plugins they should work with. These can be saved in a plugin's settings as a string.

As mentioned in PR #1911, I've started to consolidate some of the code that was replicated in individual plugins, in to MainCore - [See](https://github.com/f4exb/sdrangel/pull/1911/files#diff-bda5b7d9d1d9caa573a95ec84a33cdc709b7a0c006e52ce829c7a23ac7190b61) - There are a few issues relating to these Ids:

There's a method QString MainCore::getChannelId(const ChannelAPI* channel) to get the Id for a channel (E.g. "R2:3"). (This method potentially could be a member of ChannelAPI later on, but for now I just put all the Id related code in MainCore so it's all together). This works fine for normal channels, but I couldn't work out how it should work for MIMO channels, where there should be a stream index suffix (E.g. "M0:2.0"). It seems ChannelAPI doesn't know the stream index - only ChannelGUI does (ChannelGUI has getStreamIndex). But presumably we don't want to reference ChannelGUI in the server version of the code. 

(On a related note, one small inconvenience is that the ChannelGUI::getDeviceSetIndex() and ChannelGUI::getIndex() methods don't appear to be usable the constructor.)

A number of plugins create a list of available channels/features, and this code should probably be consolidated too. There's a bug here in several plugins, whereby if a channel with a low index is removed, the channel Id changes for the higher numbered channels, but the combo boxes still use the old Id. Also, the existing code that scans available channels doesn't generally support MIMO channels.

f4exb, 2023-12-19:
> It seems ChannelAPI doesn't know the stream index - 

Yes until recently. In order to support channels in MIMO devices in the Frequency Scanner I had to add a pure virtual method implemented in each channel instance since the stream index is part of its settings: https://github.com/f4exb/sdrangel/blob/bb406ca9a8188bc814342bbebc1ac38fe7fa65af/sdrbase/channel/channelapi.h#L156C9-L156C9

srcejon, 2024-02-16:
Have a patch that addresses most of these points.

Do we want features to be displayed as "F0:2" or just "F:2"?

f4exb, 2024-03-01:
> Do we want features to be displayed as "F0:2" or just "F:2" in comboboxes and settings?
Currently it's the former, but featureSetIndex always seems to be 0. (And is now fixed for API calls too).

Indeed there is no "Feature sets" like Device sets so the index after the "F" is meaningless hence it should rather be "F:2" in this example.

closed 2024-10-31 by srcejon