f4exb/sdrangel #1917: MAC M1 with SDRplay RSP1A

UncleBuck62, 2023-12-07:
I recently purchased an RSP1A (NOT a clone) and it works great.... for a while.

I have been testing it all and found that after a while, SDRangel just stops talking to the RSP1A and no signals are received.

Trying to stop and then restart the device in SDRangel does nothing.
Exiting the app and then starting it again also does nothing.

If I exit the app, open SDRconnect, start the device using this app, stop the device and then close SDRconnect, I can then open SDRangel again and start using the RSP1A until the next lockup.

I have tried turning on the debug log but there is nothing there to indicate there is any problem when the RSP1A stops responding.

srcejon, 2023-12-08:
What version of the API are you using?

UncleBuck62, 2023-12-08:
I am using the 3.12.1 Mac build of the API which doesn't specifically say that it is ARM and am using the Preview 2 ARM version of SDconnect.

I have been running SDRconnect for several hours now and so far there have been no lockups.

srcejon, 2023-12-08:
How long, typically, does it take to before it locks up?

Although it shouldn't be related, what demods are you using?

srcejon, 2023-12-08:
> I am using the 3.12.1 Mac build of the API which doesn't specifically say that it is ARM 

I mean which build of SDRangel? Have you tried both?

UncleBuck62, 2023-12-08:
I have only tried the ARM build of SDRangel so far.

I will download the x64 version and give that a go.

I don't really like running x64 stuff if I can avoid it, but am willing to give this a try as I really love the work you have put into SDRangel.

UncleBuck62, 2023-12-08:
I was happy to give it a shot. As I said, I really like your efforts so wanted to try it out.

Bad news though.... The x64 version refuses to see the RSP1A at all.

I went back to the ARM version and it found it straight away.

UncleBuck62, 2023-12-08:
The strange thing is that the locks appear to be random.

I have used the app for an hour or so before it locks and other times (if I don't touch anything) it can go for three or four hours before it locks.

UncleBuck62, 2023-12-08:
Forgot to say that I have been using a couple of different demods... NFM, Pager, AM and APT.

The last lockup happened after about an hour of using the app, tuning, retuning (basically just fiddling). When I stopped, I just had the device and spectrum up, walked away from the mac for about five minutes and when I came back it was locked.

UncleBuck62, 2023-12-08:
I have attached two log files.
One is when it locked up (SDRangel_with_lockup.log) at 08:49 after the app had only been running for a minute or two.

The second (SDRangel_post_lockup.log) is when I tried running the app again after the lockup and without running SDRconnect to reset the RSP1A connection.

I don't know if these will help with investigations, but thought they may.
 
[SDRangel_with_lockup.log](https://github.com/f4exb/sdrangel/files/13621235/SDRangel_with_lockup.log)
[SDRangel_post_lockup.log](https://github.com/f4exb/sdrangel/files/13621236/SDRangel_post_lockup.log)

UncleBuck62, 2023-12-08:
Not sure if this is related to the issues I am having, but the attached config crashes within seconds of clicking the "Start all devices in workspace" button.
This is not a random failure as it happens every time.

[SDRplay_NOAA.cfgx.zip](https://github.com/f4exb/sdrangel/files/13621787/SDRplay_NOAA.cfgx.zip)

UncleBuck62, 2023-12-09:
Just realised that the config file I sent you is crashing as it is to do with trying three APT decoders at the same time.

This crashing APT issue has already been raised in [#1773 ](https://github.com/f4exb/sdrangel/issues/1773) so obviously it is not the root cause of the lockups.

cfc62, 2023-12-12:
Check to see if there are any crash logs for the API.

Use the Console app, select Crash Logs from the left hand side and see if you have any 'sdr_apiService' logs.

UncleBuck62, 2023-12-12:
Nope, no crash logs for anything other than SDRangel.

UncleBuck62, 2024-01-13:
I have a feeling that this problem is specifically related to the Pager Demodulator plug-in.

I have been running the Radiosonde plug-in now for three days straight with no lockups at all but if I load the Pager Demodulator the lockups  occur again (sometimes within minutes and sometimes it takes up to about six hours).

UncleBuck62, 2024-01-14:
Sorry but I think I may have given you a bum steer on this.

I just opened up SDAngel, had no plugin-ins running and started the SDRPlay device.

Within a few minutes, the device stopped again and I was unable to restart it without first closing SDRAngel, opening SDRConnect running it, closing SDRConnect, reopening SDRAngel and then I could get the device running again.

I had the frequency set to 148.50 MHz and am now trying it at 248.50 MHz to see if the actual frequency has something to do with the problem.

Will keep you updated with further testing as I get the chance to do it.

UncleBuck62, 2024-01-14:
So I set the frequency to 248.50 MHz and started the device. I had no plugin-ins running other than the device itself and the spectrum display.

After 14 Hrs, the device was still operating with no apparent problems.

I then changed the frequency to 148.5 MHz and, in less than an hour, the device froze.

UncleBuck62, 2024-01-20:
Just an additional update on this... I decided to try running SDRConnect on the same frequency to see if it a device issue and it ran for over 24 hrs with no lockups.

Would it be possible for you to run your Duo on your MAC with the frequency set to 148.5 and see if you experience any lockups? The longest time I have been able to run it without a lockup using the RSP1A on this frequency was about 8 hrs.

UncleBuck62, 2024-01-23:
Another update.... Restarting the sdrplay_apiService also does not resolve the lockups.
Once SDRAngel (specifically the SDRPlay device) has locked, if I close SDRAngel, restart the sdrplay_apiService, restart SDRAngel and try to start the SDRPlay device, it still is locked and there is no response.

srcejon, 2024-01-23:
So you have to reboot for it to work again?

Perhaps post a screenshot showing all of your SDRangel settings.

UncleBuck62, 2024-01-23:
I don't have to reboot the mac to get it working again.
If I open SDRconnect and start the SDRplay device in that app, stop it, close SDRconnect, restart SDRAngel and the SDRPlay device will work again (for a while).

Settings screen shot attached....
[image]

srcejon, 2024-01-23:
Ok thanks. I'll see how long the Duo runs for.

srcejon, 2024-01-25:
Have run for 30+ hours without problem.

UncleBuck62, 2024-01-25:
I don't understand why it is happening either.
SDR Connect doesn't lock up at all so whatever it is must be SDRAngel specific.
Is there anything you can suggest I try to see if I can narrow down the issue.

UncleBuck62, 2024-01-25:
Were you seeing signals on the waterfall/spectrum at that frequency?
Nearly every time it has looked up here, it was whilst there was a strong signal being displayed. Only once have I seen it lock up with no signal showing.

srcejon, 2024-01-25:
Where I am, there aren't pager signals on that frequency, so no. But, I can tune to 100MHz broadcast FM and leave that running for a while.

If signals cause it to hang, that would suggest a problem in one of the demodulators rather, than the SDR driver. But I see in an earlier post you say the problem occurs even when you have no demodulators running. 

Can you press the "Show all Channels" button in the bottom toolbar in the SDRPlay device window, just to make sure nothing is hidden?

UncleBuck62, 2024-01-25:
So I tried running it with no plug-ins (confirmed there were none by clicking "Show all Channels" as requested) and it still locked up.

Stopped the device, closed SDRangel, restarted SDRangel and clicked on "Start device".

Checked in SDRangel.log and found the following lines:
`2024-01-26 00:05:45.038 (D) SDRPlayV3Input::start
2024-01-26 00:05:45.048 (C) SDRPlayV3Thread::run: sdrplay_api_Init error:  sdrplay_api_Fail
`
Not sure if that is relevant to the issue, but it looks like SDRangel is definitely the one that is blocking itself from initialising the sdr_play API somehow.

srcejon, 2024-01-25:
> Not sure if that is relevant to the issue, but it looks like SDRangel is definitely the one that is blocking itself from initialising the sdr_play API somehow.

Not entirely sure about that. SDRangel is making a call to the SDRplay API to initialise itself, and it replies by saying "Fail". I don't know of anything SDRangel can do differently. It is failing at the first step.

UncleBuck62, 2024-01-25:
In the log file there is also a warning about being unable to load library.

The line is huge and it all refers to trying to find files in "... Python3.framework/Versions/3.9/Python3"

I am running Python 3.12

Could this be related somehow?

This is the whole line.....

2024-01-26 07:26:55.847 (W) PluginManager::loadPluginsDir: Cannot load library /Applications/SDRangel.app/Contents/Resources/lib/plugins/liboutputusrp.dylib: (dlopen(/Applications/SDRangel.app/Contents/Resources/lib/plugins/liboutputusrp.dylib, 0x0085): Library not loaded: @rpath/Python3.framework/Versions/3.9/Python3
  Referenced from: <90D575CE-51AD-3E74-BDB9-776F2EFAD26A> /Applications/SDRangel.app/Contents/Frameworks/libuhd.4.5.0.dylib
  Reason: tried: '/opt/build/sdrangel/build/external/uhd/lib/Python3.framework/Versions/3.9/Python3' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/build/sdrangel/build/external/uhd/lib/Python3.framework/Versions/3.9/Python3' (no such file), '/opt/build/sdrangel/build/lib/Python3.framework/Versions/3.9/Python3' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/build/sdrangel/build/lib/Python3.framework/Versions/3.9/Python3' (no such file), '/opt/build/sdrangel/build/external/libusb/lib/Python3.framework/Versions/3.9/Python3' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/build/sdrangel/build/external/libusb/lib/Python3.framework/Versions/3.9/Python3' (no such file), '/opt/build/sdrangel/build/external/hidapi/lib/Python3.framework/Versions/3.9/Python3' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/build/sdrangel/build/external/hidapi/lib/Python3.framework/Versions/3.9/Python3' (no such file), '/opt/build/sdrangel/build/external/opus/lib/Python3.framework/Versions/3.9/Python3' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/build/sdrangel/build/external/opus/lib/Python3.framework/Versions/3.9/Python3' (no such file), '/opt/build/sdrangel/build/external/zlib/lib/Python3.framework/Versions/3.9/Python3' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/build/sdrangel/build/external/zlib/lib/Python3.framework/Versions/3.9/Python3' (no such file), '/opt/build/sdrangel/build/external/faad/lib/Python3.framework/Versions/3.9/Python3' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/build/sdrangel/build/external/faad/lib/Python3.framework/Versions/3.9/Python3' (no such file), '/opt/build/sdrangel/build/_deps/boost-build/lib/Python3.framework/Versions/3.9/Python3' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/build/sdrangel/build/_deps/boost-build/lib/Python3.framework/Versions/3.9/Python3' (no such file), '/opt/build/sdrangel/build/external/x264/lib/Python3.framework/Versions/3.9/Python3' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/build/sdrangel/build/external/x264/lib/Python3.framework/Versions/3.9/Python3' (no such file), '/opt/build/sdrangel/build/external/x265/lib/Python3.framework/Versions/3.9/Python3' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/build/sdrangel/build/external/x265/lib/Python3.framework/Versions/3.9/Python3' (no such file), '/opt/build/sdrangel/build/external/fdk-aac/lib/Python3.framework/Versions/3.9/Python3' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/build/sdrangel/build/external/fdk-aac/lib/Python3.framework/Versions/3.9/Python3' (no such file), '/opt/build/sdrangel/build/external/mpg123/lib/Python3.framework/Versions/3.9/Python3' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/build/sdrangel/build/external/mpg123/lib/Python3.framework/Versions/3.9/Python3' (no such file), '/opt/build/sdrangel/build/external/lame/lib/Python3.framework/Versions/3.9/Python3' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/build/sdrangel/build/external/lame/lib/Python3.framework/Versions/3.9/Python3' (no such file), '/opt/build/sdrangel/build/external/ffmpeg/lib/Python3.framework/Versions/3.9/Python3' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/build/sdrangel/build/external/ffmpeg/lib/Python3.framework/Versions/3.9/Python3' (no such file), '/opt/build/sdrangel/build/external/opencv/src/opencv-build/lib/Python3.framework/Versions/3.9/Python3' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/build/sdrangel/build/external/opencv/src/opencv-build/lib/Python3.framework/Versions/3.9/Python3' (no such file), '/opt/build/sdrangel/build/external/codec2/src/codec2-build/src/Python3.framework/Versions/3.9/Python3' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/build/sdrangel/build/external/codec2/src/codec2-build/src/Python3.framework/Versions/3.9/Python3' (no such file), '/opt/build/sdrangel/build/external/cm256cc/src/cm256cc-build/Python3.framework/Versions/3.9/Python3' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/build/sdrangel/build/external/cm256cc/src/cm256cc-build/Python3.framework/Versions/3.9/Python3' (no such file), '/opt/build/sdrangel/build/external/mbelib/src/mbelib-build/Python3.framework/Versions/3.9/Python3' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/build/sdrangel/build/external/mbelib/src/mbelib-build/Python3.framework/Versions/3.9/Python3' (no such file), '/opt/build/sdrangel/build/external/serialdv/src/serialdv-build/Python3.framework/Versions/3.9/Python3' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/build/sdrangel/build/external/serialdv/src/serialdv-build/Python3.framework/Versions/3.9/Python3' (no such file), '/opt/build/sdrangel/build/external/dsdcc/src/dsdcc-build/Python3.f [... cut ...] h file), '/System/Volumes/Preboot/Cryptexes/OS/opt/build/sdrangel/build/external/hackrf/src/hackrf-build/Python3.framework/Versions/3.9/Python3' (no such file), '/opt/build/sdrangel/build/external/uhd/lib/Python3.framework/Versions/3.9/Python3' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/build/sdrangel/build/external/uhd/lib/Python3.framework/Versions/3.9/Python3' (no such file), '/Applications/SDRangel.app/Contents/Frameworks/Python3.framework/Versions/3.9/Python3' (no such file))

srcejon, 2024-01-25:
That shouldn't be a problem - it's for the UHD driver.

UncleBuck62, 2024-01-27:
What about these critical errors that are in the log file?

2024-01-28 08:59:59.164 (C) SampleSinkFifo::write: (SDRPlayV3) 2306 messages dropped
2024-01-28 08:59:59.165 (C) SampleSinkFifo::write: (SDRPlayV3) overflow - dropping 1024 samples
2024-01-28 08:59:59.165 (C) SampleSinkFifo::write: (SDRPlayV3) overflow - dropping 1024 samples
2024-01-28 08:59:59.701 (C) SDRPlayV3Input::applySettings: could not set bandwidth: Rf update timed out
2024-01-28 09:00:00.333 (C) SDRPlayV3Input::applySettings: could not set IF frequency: Rf update timed out

UncleBuck62, 2024-01-28:
After a lockup, if I click "Stop all devices in workspace", then "Change device" and click "Refresh list of devices", it pauses on Enumerating srdangel.samplesource.sdrplayv3 and it eventually goes away but the SDRPlay device is no longer in the list of devices.

The following error appears in the SDRangel log file:
2024-01-28 16:16:01.335 (C) SDRPlayV3Plugin::enumOriginDevices: Failed to get SDRPlay devices:  sdrplay_api_ServiceNotResponding

If I force quit the sdrplay_api_service, it restarts, however, trying another "Refresh list of devices" produces the same problem (not in list of devices and error in log file).

There is also an entry in the sdrservice_err.log file as follows:
libusb: error [darwin_claim_interface] USBInterfaceOpen: another process has device opened for exclusive access

So it is as if SDRangel has a lock on the sdr_api_service and will not let it go, even after restarting SDRangel.

UncleBuck62, 2024-01-29:
Not 100% sure about this just yet, however, it appears this issue may be resolved.
I contacted SDRPlay as I found a frequency where SDRconnect would also lock up and they advised me to try a different USB cable as it appeared that the software was loosing connection to the SDRplay device.

I have now changed the cable and had SDRangel running for the last 4hrs with no lockups (yet).

I will leave it running for the next few hours and I will update this case with the results.

Fingers crossed.

UncleBuck62, 2024-01-30:
All appears to be working well as it has been over 15hrs now with no lockups.

I am surprised that there was no entry in any log file to indicate the cause was that the api had lost connection to the device.
Is that the sort of error that would be picked up by SDRangel or should the api be logging the error?

Any way, I am really sorry that after all this it turned out it was a harware related problem.

Thanks for all your time and efforts on this one.

This issue can now be closed.

UncleBuck62, 2024-01-30:
All resolved as it was a bad USB cable that was causing the api to loose connection to the RSP1A

closed 2024-01-30 by UncleBuck62