f4exb/sdrangel #1918: SSB demod: mitigate AGC steep transitions causing audio clicks

srs4511351, 2023-12-08:
I installed SDRangel 7.17.2-17 installed on Dec 12 2023.
Raspberry Pi 4B
SDRplay RSP1A SDR.

I appreciate your work on the AGC steep transitions causing audio clicks problem.
After your recent work on this, I am still having problems.

This is only noticeable on very strong SSB signals. Since this is a very high audio level, audio clipping is more obvious with volume set higher.
It sounds like the attack time from the initial signal is slow. Is the attack now set to the fastest possible? If not, can a control for this be added?
Activating clamping does not help, but causes distortion when they are the strongest.
Is does seem to be better, now requiring a stronger signal to cause this.

----Steve

RobRich999, 2023-12-09:
Assuming the audio is hitting hard clipping, most typical AGC configs are not likely going to act fast enough. Admittedly, I have not looked at the source code for AGC clamping, but maybe the threshold(s) could be tweaked. Otherwise a compressor and/or limiter to handle overshoots after the AGC could be a possible solution. Again, assuming it is hard clipping and not something else.

As for an potential immediate fix, curious, does your Pi have enough resources to spare for running a basic audio DSP chain? If so, consider setting the volume and AGC well below clipping in SDRangel, then use Easy Effects (or similar) to add back the desired gain and implement a single-band limiter to handle overshoots for the audio output.

https://github.com/wwmm/easyeffects

srs4511351, 2023-12-09:
Looking at the audio level meter in the Pulse Audio Volume Control, I see that when the distortion occurs, the meter peaks full scale or near it.
The normal level for these signals is much lower, somewhere below the 100% mark.
I turned the volume in SDRangel down to minimum and turned everything else up including my speaker volume, the distortion was not apparent.
I had effectively given it a lot of dynamic headroom. This shows that the distortion is from hard clipping.
Still, the initial audio peak that occurs at the start of a strong SSB signal is there, but in this case, the system can handle it without clipping.

The reason that clamping causes sustained distortion is that it limits the amount of AGC that is applied, and not enough for strong signals. Overshoot would not be a problem, as it would merely cause more AGC action, lower signal, where not enough AGC at the start of the signal is the problem.
We need AGC sooner, a faster attack, not a limiting of the effect.

EasyEffects set up for AGC with the Compressor filter did not impose a significant CPU load on my Raspberry Pi.
Turning SDRangel AGC off and the volume down, yielded AGC action with EasyEffects, but no clipping on initial peaks with attack set to .01 MS.
When I increased the attack to 100 MS, I got the same symptom as SDRangel has.
I do believe that a computer can act extremely fast and is evidenced by other SDR applications like gqrx and CubicSDR, which do not have distortion at the start of a very strong SSB signal.

----Steve

f4exb, 2023-12-09:
@srs4511351  Please feel free to submit PR to fix this issue

srs4511351, 2023-12-10:
My programming experience is relatively limited.
I am sure that this problem would best be fixed by someone familiar with this code.
However, I have looked at it. It seems that AGC isn't a simple implementation.
I want to play with the attack parameter, but...
I see that:
`int m_historySize;                      //!< Averaging length (attack)`
and
`m_historySize = historySize;`

However, I can't find where the value of historySize is assigned.

If you can help with that, I may be able to see if this can be improved.
In my opinion, any time the signal is above the threshold where AGC is to act, gain should be reduced immediately.
Analog SSB radios work similar to this and they work well.
Experimentation may prove differently.

f4exb, 2023-12-10:
OK got that. 

> int m_historySize;                      //!< Averaging length (attack)

I am not sure these comments are quite right so I will have to correct this also... in fact it is the length in samples of the moving average used to average the signal power and define the AGC value. This length is controlled with the "AGC time constant" button in power of two increments of milliseconds between 16 and 2048. It controls how fast or slow the AGC is in fact.

> In my opinion, any time the signal is above the threshold where AGC is to act, gain should be reduced immediately.

The fix I am going to post soon is doing exactly that except that the sweet spot for the AGC maximum factor value is 100 not 1. Maybe because the power target is set a a 10th of the maximum value. However I found the value experimentally not with maths.

I also removed the "clamping" code from the AGC that does not seem to do anything. Instead when switching on the "CL" button you will engage this AGC limitation.

srs4511351, 2023-12-12:
I tried the new version with the AGC changes.
On extremely strong signals, the clicking is still heard, but they seem to be shorter in duration.

f4exb, 2023-12-14:
Don´t forget it is a "mitigation". I think it is much better than before.

srs4511351, 2023-12-16:
I will lower my expectations.
It is mostly noticeable with very strong.signals.

f4exb, 2024-01-01:
Released in v7.17.3

closed 2024-01-01 by f4exb