f4exb/sdrangel #1924: High CPU% in NFM scanner mode

Axpelle, 2023-12-11:
I've observed that in scanning mode in the airband (ie AM) CPU% rarely exceeds 30-35%. However, in NFM scanning mode (marine band), CPU% shoots to 85-90%. In practical terms it's not possible to scan in NFM for any extended period. Is there a specific reason for that large difference? Can that issue be fixed? I'm using a MacBookPro running Catalina.

srcejon, 2023-12-12:
Can you attach a screenshot please?

Axpelle, 2023-12-13:
Screenshot attached
[SDRAngel Screenshot NMF scanner mode.pdf](https://github.com/f4exb/sdrangel/files/13657389/SDRAngel.Screenshot.NMF.scanner.mode.pdf)

srcejon, 2023-12-13:
Do you use the same settings for AM?

Aside from your reported issue, you have a problem in that you have device BW set to 300kHz, even though your sample rate is 2MHz. This means you are filtering a lot of the bandwidth that the frequency scanner will be trying to use (It uses the middle 75% - So 2M*.75=1.5M). You therefore need to increase the filter BW, or add some decimation to reduce the baseband bandwidth.

Axpelle, 2023-12-13:
Thank you for your help. The lowest sample rate I can get from the RSP1A is 2MHz (SDRAngel defaults back if I try to enter 1MHz when the app is off). I have increased BW to 8000Hz and set decimation at 64. The CPU% has now dropped to ±50% but is still very significant. I get the same result with the AM Demodulator.

srcejon, 2023-12-13:
> Thank you for your help. The lowest sample rate I can get from the RSP1A is 2MHz (SDRAngel defaults back if I try to enter 1MHz when the app is off).

Yep - this is what the DEC setting is for, to allow you to reduce the same rate further.

> I have increased BW to 8000Hz and set decimation at 64. 

You don't need to change both. You don't want the BW to be much wider than your SR, otherwise you aren't benefitting from it filtering out-out-band signals.

DEC at 64 will reduce the SR to 2M/64 = 31k - which is very small, so you wouldn't be able to scan many channels simultaneously.

You could try DEC=2 and BW=~1.5M

>The CPU% has now dropped to ±50% but is still very significant. I get the same result with the AM Demodulator.

What CPU do you have?

Why do you think CPU at 50% is a problem?

Axpelle, 2023-12-13:
With Dec at 2 and BW 1.536 the CPU% is around 65-68%. I need to monitor CPU performance because the MacMini I'm using has no fan and it does get very warm after about 10 min. It's a late 2014 model equipped with 1.4GHz Intel Core i5. Note that this computer is devoted exclusively to the RSP1A.

closed 2024-01-29 by github-actions[bot]