f4exb/sdrangel #1925: High CPU usage in sdrangelsrv - Is it running FFT?

wickersoft, 2023-12-12:
Hello, I'm running sdrangelsrv on a Raspberry Pi 4 with a HackRF One and a sdrangel client on the LAN. sdrangelsrv is using an unexpectedly high amount of CPU which increases proportionally to sample rate. This bottleneck limits my sample rate to 2MS/s on a regular Pi and 2.8MS/s when overclocked to 2.3GHz. The application also uses only one core. Using hackrf_transfer, I'm able to read IQ into a wav file at the full 20MS/s. Network congestion is also ruled out. 

Here's the command I'm using to start sdrangelsrv:
`sdrangelsrv --remote-tcp --remote-tcp-hwtype HackRF --fftwf-wisdom /home/pi/.config/f4exb/fftw-wisdom`

Is sdrangel running FFTs in the background or something? Is there a way to disable it that I haven't come across? I've tried disabling any spectrum views and servers using the curl API calls in the documentation, but I don't think those  are running. 

In the meantime, I'll be working on a minimal application that can serve IQ to sdrangel via this RemoteTCP/SDRA protocol. It looks pretty simple, but is the binary data documented anywhere? Thank you very much

srcejon, 2023-12-12:
> Hello, I'm running sdrangelsrv on a Raspberry Pi 4 with a HackRF One and a sdrangel client on the LAN. sdrangelsrv is using an unexpectedly high amount of CPU which increases proportionally to [...]

Try profiling it using gprof maybe?

> In the meantime, I'll be working on a minimal application that can serve IQ to sdrangel via this RemoteTCP/SDRA protocol. It looks pretty simple, but is the binary data documented anywhere? Thank you very much

This is an extension of the RTL TCP protocol.

See: https://github.com/f4exb/sdrangel/blob/master/plugins/channelrx/remotetcpsink/remotetcpprotocol.h for command IDs etc.

You might want to start with https://github.com/avian2/hackrf/blob/master/host/hackrf-tools/src/hackrf_tcp.c - which appears to be a Hack RF version of RTL TCP. So you'd need to add SDRangel's extra commands. 

But would be better if you determined if there was a problem with SDRangel's server or not...

srcejon, 2023-12-12:
Might be worth you attaching a screenshot of the client's settings.

wickersoft, 2023-12-12:
Thanks for the quick response. That should be all I need to spin up my custom server abomination. 

Unfortunately I'm not experienced with debugging applications, so I can't help you with that for now. Here's a screenshot of the remote source's settings:
[image]

If I increase the sample rate beyond 2.8M, sdrangelsrv begins to emit buffer overrun warnings from which it does not recover until I exit the process. It ceases to send any samples over the TCP connection and does not react at all to the client disconnecting. The CPU usage also doesn't go down.
[image]

srcejon, 2023-12-12:
Are you using the latest version of SDRangel? There was a bug in this code that limited sample rates, and wad only fixed recently.

wickersoft, 2023-12-12:
I compiled SDRangelSrv 7.17.2-17-g609fff1bd from source a couple of days ago. Instructions work very well!

wickersoft, 2023-12-12:
Thanks for pointing me to such useful files. I was able to modify hackrf_tcp.c to accept a connection with sdrangel and it's streaming 17MS/s without breaking a sweat. The new bottleneck seems to be the ethernet controller at 280Mbps.

closed 2024-01-28 by github-actions[bot]