f4exb/sdrangel #1930: Increase in transmit power on WFM Demodulator with a HackRF

EmilEmilchen, 2023-12-14:
I'm not 100% sure if I'm just doing something wrong but while transmitting some WFM modulated audio using SDRangel and my HackRF the I guess amplitude off the lower signals just randomly increases. When listening to the signal using an RTL SDR there is suddenly a lot off noise in the audio signal. Moving the AFBW slider of the modulator fixes the problem but then it just appears again. I attached a screenshot of when this happened. I dont think this is intended behavior and I would also be surprised if it was a hardware problem but I dont really know seing as I'm kind of a noob when it comes to RF.
[image]

wickersoft, 2023-12-17:
Are you sure your audio file is correct? Does it happen randomly or always at the same point in the file?

EmilEmilchen, 2023-12-27:
It happens on completely random occasions.

closed 2024-02-20 by github-actions[bot]