f4exb/sdrangel #1932: Crash when loading NFM demod (among other audio-related things)

agkphysics, 2023-12-19:
I've noticed that recently SDRangel will crash when loading/unloading audio-related things. The official release crashes when I try to load an NFM demod with an RTL-SDR. This is on a Windows 11 system.

So I built a debug version from source (v7.17.2) to check with a debugger and found the following call stack after trying to add an NFM demod to the RTL-SDR source:

<details><summary>Call stack</summary>
<pre>
Qt5Cored.dll!isRecursive(QMutexData * d) Line 65
	at C:\Users\qt\work\qt\qtbase\src\corelib\thread\qmutex.cpp(65)
Qt5Cored.dll!QMutex::lock() Line 232
	at C:\Users\qt\work\qt\qtbase\src\corelib\thread\qmutex.cpp(232)
sdrbase.dll!MessageQueue::push(Message * message, bool emitSignal) Line 49
	at D:\src\sdrangel\sdrbase\util\messagequeue.cpp(49)
sdrbase.dll!AudioDeviceManager::handleMessage(const Message & msg) Line 914
	at D:\src\sdrangel\sdrbase\audio\audiodevicemanager.cpp(914)
sdrbase.dll!AudioDeviceManager::handleInputMessages() Line 947
	at D:\src\sdrangel\sdrbase\audio\audiodevicemanager.cpp(947)
sdrbase.dll!AudioDeviceManager::qt_static_metacall(QObject * _o, QMetaObject::Call _c, int _id, void * * _a) Line 78
	at D:\src\sdrangel\build\sdrbase\sdrbase_autogen\PZONOMFGYT\moc_audiodevicemanager.cpp(78)
Qt5Cored.dll!QMetaCallEvent::placeMetaCall(QObject * object) Line 618
	at C:\Users\qt\work\qt\qtbase\src\corelib\kernel\qobject.cpp(618)
Qt5Cored.dll!QObject::event(QEvent * e) Line 1314
	at C:\Users\qt\work\qt\qtbase\src\corelib\kernel\qobject.cpp(1314)
Qt5Widgetsd.dll!QApplicationPrivate::notify_helper(QObject * receiver, QEvent * e) Line 3632
	at C:\Users\qt\work\qt\qtbase\src\widgets\kernel\qapplication.cpp(3632)
Qt5Widgetsd.dll!QApplication::notify(QObject * receiver, QEvent * e) Line 2972
	at C:\Users\qt\work\qt\qtbase\src\widgets\kernel\qapplication.cpp(2972)
Qt5Cored.dll!QCoreApplication::notifyInternal2(QObject * receiver, QEvent * event) Line 1063
	at C:\Users\qt\work\qt\qtbase\src\corelib\kernel\qcoreapplication.cpp(1063)
Qt5Cored.dll!QCoreApplication::sendEvent(QObject * receiver, QEvent * event) Line 1459
	at C:\Users\qt\work\qt\qtbase\src\corelib\kernel\qcoreapplication.cpp(1459)
Qt5Cored.dll!QCoreApplicationPrivate::sendPostedEvents(QObject * receiver, int event_type, QThreadData * data) Line 1817
	at C:\Users\qt\work\qt\qtbase\src\corelib\kernel\qcoreapplication.cpp(1817)
Qt5Cored.dll!QEventDispatcherWin32::sendPostedEvents() Line 1082
	at C:\Users\qt\work\qt\qtbase\src\corelib\kernel\qeventdispatcher_win.cpp(1082)
qwindowsd.dll!QWindowsGuiEventDispatcher::sendPostedEvents() Line 81
	at C:\Users\qt\work\qt\qtbase\src\platformsupport\eventdispatchers\qwindowsguieventdispatcher.cpp(81)
Qt5Cored.dll!QEventDispatcherWin32::processEvents(QFlags<enum QEventLoop::ProcessEventsFlag> flags) Line 530
	at C:\Users\qt\work\qt\qtbase\src\corelib\kernel\qeventdispatcher_win.cpp(530)
qwindowsd.dll!QWindowsGuiEventDispatcher::processEvents(QFlags<enum QEventLoop::ProcessEventsFlag> flags) Line 73
	at C:\Users\qt\work\qt\qtbase\src\platformsupport\eventdispatchers\qwindowsguieventdispatcher.cpp(73)
Qt5Cored.dll!QEventLoop::processEvents(QFlags<enum QEventLoop::ProcessEventsFlag> flags) Line 140
	at C:\Users\qt\work\qt\qtbase\src\corelib\kernel\qeventloop.cpp(140)
Qt5Cored.dll!QEventLoop::exec(QFlags<enum QEventLoop::ProcessEventsFlag> flags) Line 232
	at C:\Users\qt\work\qt\qtbase\src\corelib\kernel\qeventloop.cpp(232)
Qt5Widgetsd.dll!QDialog::exec() Line 602
	at C:\Users\qt\work\qt\qtbase\src\widgets\dialogs\qdialog.cpp(602)
sdrgui.dll!DeviceGUI::openAddChannelsDialog() Line 384
	at D:\src\sdrangel\sdrgui\device\devicegui.cpp(384)
sdrgui.dll!DeviceGUI::qt_static_metacall(QObject * _o, QMetaObject::Call _c, int _id, void * * _a) Line 164
	at D:\src\sdrangel\build\sdrgui\sdrgui_autogen\GULY2SVRVA\moc_devicegui.cpp(164)
Qt5Cored.dll!doActivate<0>(QObject * sender, int signal_index, void * * argv) Line 3901
	at C:\Users\qt\work\qt\qtbase\src\corelib\kernel\qobject.cpp(3901)
Qt5Cored.dll!QMetaObject::activate(QObject * sender, const QMetaObject * m, int local_signal_index, void * * argv) Line 3947
	at C:\Users\qt\work\qt\qtbase\src\corelib\kernel\qobject.cpp(3947)
Qt5Widgetsd.dll!QAbstractButton::clicked(bool _t1) Line 309
	at C:\Users\qt\work\qt\qtbase\src\widgets\.moc\debug\moc_qabstractbutton.cpp(309)
Qt5Widgetsd.dll!QAbstractButtonPrivate::emitClicked() Line 417
	at C:\Users\qt\work\qt\qtbase\src\widgets\widgets\qabstractbutton.cpp(417)
Qt5Widgetsd.dll!QAbstractButtonPrivate::click() Line 409
	at C:\Users\qt\work\qt\qtbase\src\widgets\widgets\qabstractbutton.cpp(409)
Qt5Widgetsd.dll!QAbstractButton::mouseReleaseEvent(QMouseEvent * e) Line 1045
	at C:\Users\qt\work\qt\qtbase\src\widgets\widgets\qabstractbutton.cpp(1045)
Qt5Widgetsd.dll!QWidget::event(QEvent * event) Line 8646
	at C:\Users\qt\work\qt\qtbase\src\widgets\kernel\qwidget.cpp(8646)
Qt5Widgetsd.dll!QAbstractButton::event(QEvent * e) Line 1002
	at C:\Users\qt\work\qt\qtbase\src\widgets\widgets\qabstractbutton.cpp(1002)
Qt5Widgetsd.dll!QPushButton::event(QEvent * e) Line 696
	at C:\Users\qt\work\qt\qtbase\src\widgets\widgets\qpushbutton.cpp(696)
Qt5Widgetsd.dll!QApplicationPrivate::notify_helper(QObject * receiver, QEvent * e) Line 3632
	at C:\Users\qt\work\qt\qtbase\src\widgets\kernel\qapplication.cpp(3632)
[... 69 lines cut ...]
{
    quintptr u = quintptr(d);
    if (Q_LIKELY(u <= 0x3))
        return false;
#ifdef QT_LINUX_FUTEX
    Q_ASSERT(d->recursive);
    return true;
#else
    return d->recursive; // Immediate cause of exception
#endif
}
```

Caused by:
https://github.com/f4exb/sdrangel/blob/bb406ca9a8188bc814342bbebc1ac38fe7fa65af/sdrbase/util/messagequeue.cpp#L48
In turn caused by:
https://github.com/f4exb/sdrangel/blob/bb406ca9a8188bc814342bbebc1ac38fe7fa65af/sdrbase/audio/audiodevicemanager.cpp#L913

I also noted there was a similar crash when I tried to unload an audio input source, without first starting aquisition, caused by:
https://github.com/f4exb/sdrangel/blob/bb406ca9a8188bc814342bbebc1ac38fe7fa65af/sdrbase/audio/audiofifo.cpp#L84

And also when stopping acquisition, caused by:
https://github.com/f4exb/sdrangel/blob/bb406ca9a8188bc814342bbebc1ac38fe7fa65af/plugins/samplesource/audioinput/audioinput.cpp#L143

Since all of these involve mutexes and audio I'm not sure whether it's an issue with SDRangel or my audio configuration.

srcejon, 2023-12-22:
What audio devices do you have? A screenshot might be helpful.

agkphysics, 2023-12-22:
Here are my input and output audio devices:

[image]
[image]

The default output is Realtek HD Audio 2nd output and the default input device is the webcam microphone.

agkphysics, 2024-02-05:
I have tried to do some more debugging but have not found the root cause, either in SDRangel or externally.

However, the most recent update (v7.12.3) seems to have fixed the issue. I will update if I notice the crash again.

closed 2024-02-05 by agkphysics