f4exb/sdrangel #1933: ADALM Pluto support in MacOS DMG

tomtastic, 2023-12-19:
This doesn't appear to be enabled currently in the latest MacOS builds ?

I would like to see all of this in the MacOS builds if possible? :
* `libiio` enabled
* ADALM Pluto support enabled
* The `plutosdrmimo` plugin enabled

srcejon, 2023-12-22:
Last time I tried, including libiio was problematic.

tomtastic, 2023-12-22:
So I had a look at building manually on MacOS (Ventura 13.6.1), as it seems that as `libiio` should build by default, so the ADALM Pluto should be supported without doing anything special.

Whilst doing that I discovered :
### Issue 1 - LAME
Of the external libraries that get pulled down during the build, `LAME` was failing to build it's frontend binaries (which we dont need) due to:
```C
parse.c:417:34: error: call to undeclared function 'id3tag_set_comment_ucs2'; ISO C99 and later do not support implicit function declarations [-Wimplicit-function-declaration]
```
So I made a small change to the configure parameters to disable building the LAME frontend binaries :
https://github.com/f4exb/sdrangel/blob/bb406ca9a8188bc814342bbebc1ac38fe7fa65af/external/CMakeLists.txt#L486

I've created a PR for this `LAME` frontend build bug, though it's perhaps unrelated to why `libiio` doesn't appear in the current MacOS releases?

### Issue 2 - UHD
Then I hit a problem building UHD (for USRP?) due to:
```C
/tmp/sdrangel-20231220-29072-lc7i06/sdrangel-7.17.2/build/external/uhd/src/uhd/host/lib/usrp/mpmd/sim_find.cpp:13:10: fatal error: 'Python.h' file not found
```
I've no idea why this happens, as we're passing in `-DPYTHON_EXECUTABLE=/opt/homebrew/bin/python3` to cmake, and I can see the compilation added the appropriate include path `-I/opt/homebrew/Frameworks/Python.framework/Versions/3.11/include/python3.11`, which I confirmed contains the expected `Python.h`.
To workaround this, I added `-DENABLE_USRP=OFF`.
Is this related to UHD missing as reported here https://github.com/f4exb/sdrangel/issues/1870 ?

### Results
Though I can now successfully build (around 1h10mins on my M1 MBP), I am struggling to get cmake to set my the rpath appropriately for the installed location so it can find the shared libraries it's built.
I think mostly a success though?

srcejon, 2023-12-22:
Yeah, it's not building libiio that's the problem, it's getting a packaged build that runs correctly.

tomtastic, 2023-12-22:
I have a lot to learn yet about cmake, but I'm hoping I can figure out the rpath business.

I'm not aiming for a packaged build, but rather a Homebrew formula something like  :
```ruby
class Sdrangel < Formula
  desc "SDR Rx/Tx software for Airspy, Airspy HF+, BladeRF, HackRF, LimeSDR, PlutoSDR, RTL-SDR, SDRplay RSP1 and FunCube"
  homepage ""
  url "https://github.com/f4exb/sdrangel/archive/refs/tags/v7.17.2.tar.gz"
  sha256 "ad24bfdd19a641de38afeaae8bca2c8dccdd7229573535070d2a120583524d18"
  license "GPL-3.0"

  depends_on "cmake" => :build
  depends_on "python"
  depends_on "qt5"

  def install
    ENV.deparallelize  # the formula will fails if building in parallel
    # ENABLE_EXTERNAL_LIBRARIES (ON) - We want libiio, etc to be pulled down and built!
    # ENABLE_USRP (OFF) - /build/external/uhd/src/uhd/host/lib/usrp/mpmd/sim_find.cpp:13:10: fatal error: 'Python.h' file not found
    system "cmake", "-S", ".", "-B", "build", *std_cmake_args, "-DENABLE_EXTERNAL_LIBRARIES=ON", "-DPYTHON_EXECUTABLE=/opt/homebrew/bin/python3", "-DENABLE_USRP=OFF", "-DBUNDLE=ON", "-DCMAKE_BUILD_TYPE=Release", "-DCMAKE_MACOSX_RPATH=1", "-DCMAKE_BUILD_WITH_INSTALL_RPATH=1"
    system "cmake", "--build", "build"
    system "cmake", "--install", "build"
  end

  test do
    # `test do` will create, run in and delete a temporary directory.
    #
    # This test will fail and we won't accept that! For Homebrew/homebrew-core
    # this will need to be a test that verifies the functionality of the
    # software. Run the test with `brew test sdrangel`. Options passed
    # to `brew install` such as `--HEAD` also need to be provided to `brew test`.
    #
    # The installed folder is not in the path, so use the entire path to any
    # executables being tested: `system "#{bin}/program", "do", "something"`.
    system "false"
  end
end
```

srcejon, 2024-01-04:
Just uploaded a new arm64 dmg that should include Pluto drivers. I don't have one to test it with, so let me know how you get on.

tomtastic, 2024-01-08:
Thank you. 
I can confirm that I can see the Pluto plugins are there and load OK, and I can add a Pluto Rx device,
however I'm away on vacation still and dont have my actual SDR to test with.

kda186, 2024-02-02:
> 刚刚上传了一个新的 arm64 dmg，其中应包含 Pluto 驱动程序和 libiio。我没有可以测试的东西，所以请告诉我你的进展如何。

Where can I get this "arm64 dmg, which should contain Pluto driver and libiio" I am using 7.17.3-1539-g5dcf7f7d4 and there is no pluto option in the form list, thank you

Lew-Engineering, 2024-02-21:
I installed the latest sdrangel-7.18.0-macx64.dmg  on and Intel X64 Mac and the Pluto SDR is not in the list of selectable devices.  Is there a way to check if this build supports the Pluto SDR on X64 Mac?

srcejon, 2024-02-21:
> I installed the latest sdrangel-7.18.0-macx64.dmg on and Intel X64 Mac and the Pluto SDR is not in the list of selectable devices. Is there a way to check if this build supports the Pluto SDR on X64 Mac?

It's not currently in the X86 build. You'd have to compile it yourself.

srcejon, 2024-02-21:
7.18.1 x86 build should have pluto support included - but I don't have one, so can't test if it works or not.

Lew-Engineering, 2024-02-22:
7.18.1 works perfectly on my x86 Mac and the Pluto.  Thank you!

srcejon, 2024-02-23:
Should now be supported on both x64 and arm64.

closed 2024-02-23 by srcejon

tomtastic, 2024-02-23:
Thanks @srcejon, I did have a play with it last week (no free time :<), and it seemed to work well,
despite me not having any luck getting any sound of it, but thats probably my issue.