f4exb/sdrangel #1934: SOAPYremote not in device list (v7.17.2 OSX)

kobold81, 2023-12-21:
Hello,

as far as I understand, sdrangel should be able to conect to a soapyremote instance.
"Select samplin device" does not show an entry for soapy in v7.17.2 for OSX Version. Do I have to compile sdrangel or will future builds include sopy?

srcejon, 2023-12-22:
Soapy isn't included in the Mac build currently, so you'd need to compile it.

Note also, you need to use command line option --soapy to enable it and to connect to remote devices, see: https://github.com/f4exb/sdrangel/wiki/Soapy-Remote

kobold81, 2023-12-22:
thanks, is it included in the other OS builds?

f4exb, 2023-12-27:
Sorry Soapy will remain optional

srcejon, 2024-01-04:
The latest macarm64 dmg should contain Soapy Remote (but no other Soapy drivers). Untested, so let me know how you get on.

kobold81, 2024-01-04:
Hi, 
I own a intel mac, so if I have time in the near future, I will tr an build sdrangel with soapy addons :-)

closed 2024-02-19 by github-actions[bot]