f4exb/sdrangel #1939: AIS plugin crash on Android

BrunoKlu, 2023-12-26:
Hello, I'm having this crash on the last update, not sure where this is coming from.

Here is the phones crash log : 

```
type: crash
osVersion: google/panther/panther:14/UQ1A.231205.015/2023121200:user/release-keys
package: org.sdrangel:10
process: org.sdrangel
processUptime: 0 + 0 ms
installer: com.android.vending

signal 11 (SIGSEGV), code 2 (SEGV_ACCERR), fault addr 0xb400cdbccde2affc

backtrace:
      #00 pc 000000000002d55c  /data/app/~~skQoJ-sYwp9Mye12hRIfEw==/org.sdrangel-WX_ESZfiWbyqiCbJFg9MpA==/lib/arm64/libdemodais_arm64-v8a.so (AISDemodSink::processOneSample(std::__ndk1::complex<float>&)+1200) (BuildId: 23553e89f61815ee82057bb983915ee22d9673d6)
      #01 pc 000000000002cd90  /data/app/~~skQoJ-sYwp9Mye12hRIfEw==/org.sdrangel-WX_ESZfiWbyqiCbJFg9MpA==/lib/arm64/libdemodais_arm64-v8a.so (AISDemodSink::feed(std::__ndk1::__wrap_iter<Sample const*> const&, std::__ndk1::__wrap_iter<Sample const*> const&)+108) (BuildId: 23553e89f61815ee82057bb983915ee22d9673d6)
      #02 pc 000000000002aed8  /data/app/~~skQoJ-sYwp9Mye12hRIfEw==/org.sdrangel-WX_ESZfiWbyqiCbJFg9MpA==/lib/arm64/libdemodais_arm64-v8a.so (AISDemodBaseband::handleData()+200) (BuildId: 23553e89f61815ee82057bb983915ee22d9673d6)
      #03 pc 0000000000483e7c  /data/app/~~skQoJ-sYwp9Mye12hRIfEw==/org.sdrangel-WX_ESZfiWbyqiCbJFg9MpA==/lib/arm64/libQt5Core_arm64-v8a.so (QObject::event(QEvent*)+456) (BuildId: 1f19d00d68ade81c7dfcfcae973f3fadf4d3ef32)
      #04 pc 000000000026efe4  /data/app/~~skQoJ-sYwp9Mye12hRIfEw==/org.sdrangel-WX_ESZfiWbyqiCbJFg9MpA==/lib/arm64/libQt5Widgets_arm64-v8a.so (QApplicationPrivate::notify_helper(QObject*, QEvent*)+296) (BuildId: 9b49e193620cac8290676ecd3d5378fad8fc5c08)
      #05 pc 0000000000270404  /data/app/~~skQoJ-sYwp9Mye12hRIfEw==/org.sdrangel-WX_ESZfiWbyqiCbJFg9MpA==/lib/arm64/libQt5Widgets_arm64-v8a.so (QApplication::notify(QObject*, QEvent*)+560) (BuildId: 9b49e193620cac8290676ecd3d5378fad8fc5c08)
      #06 pc 000000000046a2f8  /data/app/~~skQoJ-sYwp9Mye12hRIfEw==/org.sdrangel-WX_ESZfiWbyqiCbJFg9MpA==/lib/arm64/libQt5Core_arm64-v8a.so (QCoreApplication::notifyInternal2(QObject*, QEvent*)+176) (BuildId: 1f19d00d68ade81c7dfcfcae973f3fadf4d3ef32)
      #07 pc 000000000046afb0  /data/app/~~skQoJ-sYwp9Mye12hRIfEw==/org.sdrangel-WX_ESZfiWbyqiCbJFg9MpA==/lib/arm64/libQt5Core_arm64-v8a.so (QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*)+432) (BuildId: 1f19d00d68ade81c7dfcfcae973f3fadf4d3ef32)
      #08 pc 00000000004a018c  /data/app/~~skQoJ-sYwp9Mye12hRIfEw==/org.sdrangel-WX_ESZfiWbyqiCbJFg9MpA==/lib/arm64/libQt5Core_arm64-v8a.so (QEventDispatcherUNIX::processEvents(QFlags<QEventLoop::ProcessEventsFlag>)+76) (BuildId: 1f19d00d68ade81c7dfcfcae973f3fadf4d3ef32)
      #09 pc 0000000000468120  /data/app/~~skQoJ-sYwp9Mye12hRIfEw==/org.sdrangel-WX_ESZfiWbyqiCbJFg9MpA==/lib/arm64/libQt5Core_arm64-v8a.so (QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>)+356) (BuildId: 1f19d00d68ade81c7dfcfcae973f3fadf4d3ef32)
      #10 pc 0000000000348818  /data/app/~~skQoJ-sYwp9Mye12hRIfEw==/org.sdrangel-WX_ESZfiWbyqiCbJFg9MpA==/lib/arm64/libQt5Core_arm64-v8a.so (QThread::exec()+96) (BuildId: 1f19d00d68ade81c7dfcfcae973f3fadf4d3ef32)
      #11 pc 0000000000349580  /data/app/~~skQoJ-sYwp9Mye12hRIfEw==/org.sdrangel-WX_ESZfiWbyqiCbJFg9MpA==/lib/arm64/libQt5Core_arm64-v8a.so (BuildId: 1f19d00d68ade81c7dfcfcae973f3fadf4d3ef32)
      #12 pc 00000000000cf9ec  /apex/com.android.runtime/lib64/bionic/libc.so (__pthread_start(void*)+204) (BuildId: 118bbf54d490f2f7657459db9db7372f)
      #13 pc 0000000000064730  /apex/com.android.runtime/lib64/bionic/libc.so (__start_thread+64) (BuildId: 118bbf54d490f2f7657459db9db7372f)

```

Thanks 🙏

srcejon, 2023-12-26:
Runs OK here.

Can you perhaps describe what happens in more detail? Does it happen immediately? Rarely? When specific data is received?

What other demods / features do you have open?

Screenshots / IQ .wavs might be useful.

BrunoKlu, 2023-12-27:
Hello,

Well I just Open the AIS config on startup. Then when I hit play it crashes immediately. I'm using RTL SDR Blog V4 the phones running Graphene OS.
This is My Hiking go kit so will try to upload more infos tonight at next camp.as I'm on a hike for 2 weeks. Thanks for the reply.

closed 2024-02-12 by github-actions[bot]