f4exb/sdrangel #1941: Feature request: global save function

alibenpeng, 2023-12-29:
I love sdrangel for its features and versatility, but at least on my computer it crashes quite often. That wouldn't normally be something I couldn't live with, but the problem is, that in a crash, sdrangel doesn't save its settings, including all the configurations and presets I've saved in that session. It would be a great relief to be able to just press ctrl-s every once in a while and not lose all the changes to a session when something bad happens.

alibenpeng, 2023-12-29:
never mind. I just found it. Anyhow, a keyboard shortcut like ctrl-s would still be nice, though.

closed 2023-12-29 by alibenpeng

srcejon, 2023-12-30:
You might want to create an issue about what you were using when it crashes. Please include screenshots and all other details you can think of.