f4exb/sdrangel #1942: Sound problems, as always under Unbuntu, but unfortunately SDRAngel too

cortmen, 2024-01-02:
Hello, perhaps someone has the crucial tip.  
 
Given :Unbuntu 22.04 ,x86_64, 2 Core, 8GB, & Sound  Realtek ALC5662-CG and RTL-SDR Blog V4

I have recompiled SDRAngel, without major errors, the instructions in the wiki are perfect.  

Unfortunately I just can't get any sound under SDRAngel.  Desktop, system sounds, VLC, Firefox (youtube etc) all
apps have sound.

The small programme pavucontroll shows that sound is also transmitted on the HDMI1 (default device) connection. Unfortunately, only SDRAngel remains muted and is not displayed in the application under pavucontrol.
Calling up SDRAngel as root, no sound.

I have installed the master-tar.gz binary, makes no difference. No sound.

Debug - log 
``
AudioOutputDevice::start: device: -1 rate: 8000 thread: 0x55e3dbe39630
2024-01-02 19:00:29.496 (W) AudioOutputDevice::start: using system default device 
2024-01-02 19:00:29.497 (W) AudioOutputDevice::start: format 8000 Hz 2xS16LE audio/pcm not supported. Using:  sampleRate: -1 channelCount: -1 sampleSize: -1 codec:  byteOrder: LE sampleType: 0
2024-01-02 19:00:29.497 (W) AudioOutputDevice::start: Audio device '' `failed`

srcejon, 2024-01-02:
What demod are you using?

What other audio devices do you have if you right click on the audio mute icon in the demod settings?

cortmen, 2024-01-02:
Thank you for your help.
Demodulator AM and also Broadcast FM, squelch open, Volume 100%

I have just connected a SoundBlaster USB.
It was recognized correctly under Unbuntu.
pavccontrol shows HDMI1 (default device) and USB Digital / Analog Sound. Even if I switch off the default device (HDMI1) and set USB Audio as default, this does not change anything under SDR Angel.

Why does it always say Default Device or Default Input device under Audio?  Does SDR Angel only use the Default Audio device?

srcejon, 2024-01-02:
>Why does SDRAngel always say "Default Device"or "Default Input device" under Audio

Are you using Preferences > Audio or right clicking on the Audio Mute button in the AM Demod?

cortmen, 2024-01-02:
In both cases, only default Device

closed 2024-02-17 by github-actions[bot]