f4exb/sdrangel #1943: AIS message type 5  in AIS modulator

Cubster1990, 2024-01-03:
Hi 
Got the modulator working with UDP and the API to TX type 1 messages "Scheduled position report" but as I understand AIS RX stations also need a type 5 message "Static and voyage related data". 
The type 5 message includes static info such as: IMO, Vessel name, Call sign, Ship type, ETA, Draught and Destination. 

How would I transmit a type 5 message in SDRangel?

srcejon, 2024-01-03:
You'd need to encode it yourself as hex in the Msg field.

Cubster1990, 2024-01-29:
> You'd need to encode it yourself as hex in the Msg field.

Been reading up on the encoding from [this PDF](https://www.itu.int/dms_pubrec/itu-r/rec/m/R-REC-M.1371-5-201402-I!!PDF-E.pdf) from this earlier  [post](https://github.com/f4exb/sdrangel/issues/1927) .
Haven't been able to "crack" the encoding from NMEA to HEX in order to generate my own type 5 message.
Any suggestions?

srcejon, 2024-01-29:
You shouldn't need NMEA. Create the message a sequence of bits according to the table, then convert to hex.

closed 2024-03-16 by github-actions[bot]