f4exb/sdrangel #1945: Request ELAD TM-2

ab1x, 2024-01-04:
I really like sdrangel, it is a great app for SDR enthusiasts. It has many features and supports a wide range of hardware. I especially appreciate the 2D and 3D visualization of the signals, and the satellite tracking function. I hope that in the future, sdrangel will support ELAD TM-2, which is a very convenient and ergonomic controller for SDR software. It would be awesome to use sdrangel with TM-2 and enjoy the best of both worlds. Thank you for your hard work and dedication to this project.

srcejon, 2024-01-04:
Have you seen the Jogdial feature: https://github.com/f4exb/sdrangel/blob/master/plugins/feature/jogdialcontroller/readme.md

If it can emulate keyboard events, it might work

ab1x, 2024-01-04:
Hello, thank you for your suggestion to try Jogdial feature with sdrangel and elad tm-2. I appreciate your help. I have tried it but it did not work for me. Do you have any idea how to make it work? If you do, I would be happy to try it. Thank you again for your assistance.

أ

srcejon, 2024-01-04:
I don't have have an Elad TM-2, so you'd need to tell us exactly what it can do and what you tried.

closed 2024-02-19 by github-actions[bot]