f4exb/sdrangel #1947: Frequency scanner  scan DMR

autodiag, 2024-01-06:
despite all the efforts and reading the manual several times, I am not able to figure out how to scan DMR signals..for example I have three DMR frequencies in the frequency scanner that I want to scan (scanner is connected to DSD demodulator) ..the problem is that if the scanner is on the iddling frequency, it is on data and the other channels are not available, it seems to me as if it detects only data frame instead of voice frame
maybe I'm doing something wrong,if so I'm not sure what...sacnner in DMR mode should only respond to voice frame,whether free or encoded and not to data or iddling of the channel

srcejon, 2024-01-06:
The freq scanner only works on power levels - it doesn't understand a difference between voice and data.

autodiag, 2024-01-06:
thanks for the reply, I thought so
Will it be possible to add this feature to the scan ?
nowadays when there are so many DMR networks everywhere, I think it's almost a necessity.

srcejon, 2026-09-20:
We can try to use slot1On and slot2On in the DMR's report.

srcejon, 2026-11-12:
Should be supported in v8.2

closed 2026-11-12 by srcejon