f4exb/sdrangel #1948: ability to manually key morse in ssb

internetuser22, 2024-01-08:
suggestion ability to manually key with a straight key, by com port or mouse

f4exb, 2024-01-08:
These controls do exist: check under the "WPM" button (it is not in the main screenshot) Look at 16.7 to 16.10 under here https://github.com/f4exb/sdrangel/tree/master/plugins/channeltx/modssb#166-text-playstop 

To use a hardware key you will need another program to be able to generate keystrokes.

f4exb, 2024-01-15:
It turns out that the easier way to interface a morse key to a computer is to recycle a mouse by connecting the right and left buttons to the iambic keys (for iambic) or just one of them for a straight key. Mouse control is not implemented and it would be an improvement if was.

More importantly a sound feedback should be linked to the keys. It is not possible to use the radio feedback because of the delay.

closed 2024-03-13 by f4exb

f4exb, 2024-03-13:
With the present design of audio control it does not seem possible to avoid a slight delay on audio. So even when activating the audio feedback this delay is still annoying when keying directly with the mouse. Therefore I think a CW keyer with a buzzer for audio feedback is necessary while using the straight key mode in the SSB modulator.

commit b0b4fa52 (2024-03-13): Use mouse keys for CW modulation in SSB modulator. Fixes #1948