f4exb/sdrangel #1953: SDRAngel ARM MacOS from sdrangel-7.17.3-macarm64.dmg do not run

RubbaCippi, 2024-01-14:
1) download the sdrangel-7.17.3-macarm64.dmg
2) open the dmg
3) observe that the app has a “forbid” sign on it
4) double click on the app and read the alert windows (this app don’t run on this system: trash or update)

MacBook Pro 14” M1max
macOS Ventura 13.5.2

[Screenshot 2024-01-14 alle 23.01.23.pdf](https://github.com/f4exb/sdrangel/files/13933259/Screenshot.2024-01-14.alle.23.01.23.pdf)
[Screenshot 2024-01-14 alle 23.02.30.pdf](https://github.com/f4exb/sdrangel/files/13933260/Screenshot.2024-01-14.alle.23.02.30.pdf)

hambertrand, 2024-01-16:
Same issue here... Up until I updated to MacOS 14.2.1 last night. SDRAngel now opens and runs fine.

RubbaCippi, 2024-01-16:
Interesting.
But for the moment I don’t like to update to Sonoma.

srcejon, 2024-01-16:
Don't know what the problem is - but it was built on Ventura 13.6.1.

RubbaCippi, 2024-01-16:
Ah, ok!
I am on Ventura 13.5.2 and the presence of Sonoma between the updates available masked 13.6.3
Now I apply the update.

Thank You!

srcejon, 2024-01-17:
Have updated the Wiki to indicate Ventura 13.6.1 or newer is required. Thanks.

closed 2024-01-17 by srcejon